pimoroni/inky #4: update only part of the screen?

pootle, 2018-11-07:
Is it possible to update only part of the screen?

The inky interface only seems to allow entire screen refresh. Would it be possible to update only part of the screen - even if this meant not using deep sleep this might make small updates a lot faster.

Gadgetoid, 2018-11-12:
There's no hardware support for partial refresh, so unfortunately this is impossible. On bi-colour displays it's basically useless anyway as the way the display works is fuzzy enough that it distorts pixels around the partial update area (we saw this on our original prototypes with a different display part).

closed 2018-11-12 by Gadgetoid

MattiJarvinen-BA, 2019-02-19:
Damn... Thanks for the info.

chrislast, 2019-02-26:
The mono version at https://www.china-epaper.com/uploads/soft/DEPG0420A01V3.0.pdf suggests that SPI command 0x22 0xC0 will do a partial update in Section 13.4 even though it doesn't show it as supported in the command table.  The multicolour manual at https://www.china-epaper.com/uploads/soft/DEPG0420R01V3.0.pdf doesn't even suggest it although it does still support the partial ram address commands 0x44, 0x45, 0x4e,  0x4f as well so it would probably work, possibly badly for the reasons Gadgetoid mentioned.

It looks like you'd have to keep the EPD powered up between partial refreshes to work correctly so you'll sacrifice some current.   Not sure whether gadgetoid kept it powered when testing...

Anyway I came here wondering the same thing so now I know thanks for the answer :)

STrRedWolf, 2022-04-12:
Just saw this and wondering if the Inky Impression was able to partially update.  Different hardware and all.

Gadgetoid, 2022-04-13:
Impression also cannot do partial updates for largely the same reasons. Frustrating because the update cycle is so long that you lose a bunch of ambient information for almost half a minute.

juj, 2024-03-27:
> There's no hardware support for partial refresh, so unfortunately this is impossible. On bi-colour displays it's basically useless anyway as the way the display works is fuzzy enough that it distorts pixels around the partial update area

This statement seems a bit self-contradictory, no? First stating there is no partial refresh support, but then saying that when doing (such nonexistent?) partial refresh, it was observed to distort pixels around the partial update area? I wonder if the intent was to write "there's no software support for partial refresh", or I wonder if these statements were referring to different boards, or something like that?

Btw, even if partial update would be distorting pixels around the partial update area, maybe it could still be useful for some use cases? e.g. for UI HUDs that have distinct sections for different data, and blank white/black area between those sections?

It looks like in the library at least there is code that submits an update rectangle: https://github.com/pimoroni/inky/blob/6162b5fbc2eba8b7a56b41eb03e4438db9bea6eb/library/inky/inky.py#L301-L308

which is currently hardcoded to send a full update rectangle. Maybe changing that might be something to play with? Are those fields honored by any of the display controllers?

Thanks btw for diligently explaining the LUTs at https://github.com/pimoroni/inky/blob/6162b5fbc2eba8b7a56b41eb03e4438db9bea6eb/library/inky/inky.py#L111-L157 and https://github.com/pimoroni/inky/issues/13#issuecomment-456814320 . That is very helpful information. (also good information at https://forums.pimoroni.com/t/my-inky-phat-clock-refresh-speed-question/6955/5 )

Gadgetoid, 2024-03-28:
> This statement seems a bit self-contradictory, no?

Not really, I can assert that cats can't fly but I still have a very thorough understanding of the drawbacks that might occur if they could 😆

The technology in e-ink is not especially varied, though I'll admit I haven't *tried* to perform a partial update since those underwhelming attempts with some of our first products- so maybe things have improved!

> maybe it could still be useful for some use cases?

You'd be forgiven for thinking so, but it really does kinda suck. It's also a balancing act between usefulness and the time investment to get it up and running, update the libraries, do releases and so on for the - *looks up at replies to this issue from 2018* - I think it's fair to say handful of people who might use it.

See also: https://github.com/pimoroni/inky/issues/138

I welcome anyone to emphatically prove me wrong, invest the time and do something useful with it. 

Here's a fantastic example of someone absolutely *schooling* me on the finer points of e-ink - https://github.com/antirez/uc8151_micropython