pimoroni/inky #13: LUT info, and possibility to clear with black?

mattdesl, 2018-12-22:
Hey there! Having fun tinkering with my inky pHAT! I've been adding new lookup tables to get a [custom update sequence](https://twitter.com/mattdesl/status/1075869530793095169) and wondered:

One step in the sequence is "clear the buffer" which makes the screen go entirely white. Is there a similar command or hex code that would clear with *black* instead of clearing with white?

Currently my sequence does this:

1. Draw screen entirely with black or white pixels depending on next image, then run busy_wait
2. Now draw screen with a custom LUT depending on whether we are drawing in from black or white, run busy_wait again

I'd like to remove step 1 from this sequence to speed things up. To do this, I need a way to clear the screen with white *or* black.

It actually already seems possible to clear to white using `0b10100000` in the LUT, which could potentially remove step 1 in the context of a white background. However, another issue #12 is preventing me from doing that, as for whatever reason the clear operation is resetting the border to black.

Any ideas? :smile: 

Thanks!

Gadgetoid, 2019-01-23:
The LUTs have always managed to just hover at the very fringes of my understanding, but some wisdom from the original Inky pHAT library may help.

The codes in the phase tables of the LUTs represent the different positive and negative voltages set on each step. The voltage that affects the ink particulate is the difference between Black and VCOM or White and VCOM. Generally any pair of codes that pull the ink in *one* direction will, when reversed, pull it in the other.

On power on these are the bits and voltages they correspond to:

```
VSS  = 0b00 = Ground
VSH1 = 0b01 = 15V
VSL  = 0b10 = -15V
VSH2 = 0b11 = 5.4V
```

But these voltages can be configured and, indeed, are changed for the Yellow LUTs in this code.

You'll notice patterns like `0b01000100` across the A, B, C and D steps of each Phase. This pulses the ink in the desired direction, followed by a settling period. The numbers in the duration table below the phase LUTs correspond to how long each pulse will last, and the "Repeat" number is how many times that particular phase should be repeated.

Now taking our default black LUTs I can explain - more or less - what's happening:

```python
            'black': [
            # Phase 0     Phase 1     Phase 2     Phase 3     Phase 4     Phase 5     Phase 6
            # A B C D     A B C D     A B C D     A B C D     A B C D     A B C D     A B C D
            0b01001000, 0b10100000, 0b00010000, 0b00010000, 0b00010011, 0b00000000, 0b00000000,  # LUT0 - Black
            0b01001000, 0b10100000, 0b10000000, 0b00000000, 0b00000011, 0b00000000, 0b00000000,  # LUTT1 - White
            0b00000000, 0b00000000, 0b00000000, 0b00000000, 0b00000000, 0b00000000, 0b00000000,  # IGNORE
            0b01001000, 0b10100101, 0b00000000, 0b10111011, 0b00000000, 0b00000000, 0b00000000,  # LUT3 - Red
            0b00000000, 0b00000000, 0b00000000, 0b00000000, 0b00000000, 0b00000000, 0b00000000,  # LUT4 - VCOM

            # Duration            |  Repeat
            # A   B     C     D   |
            16,   4,    4,    4,     4,   # 0 Flash
            16,   4,    4,    4,     4,   # 1 clear
            4,    8,    8,    16,    16,  # 2 bring in the black
            0,    0,    0,    0,     0,   # 3 time for red
            0,    0,    0,    0,     0,   # 4 final black sharpen phase
            0,    0,    0,    0,     0,   # 5
            0,    0,    0,    0,     0,   # 6
],
```

## Phase 0 - Flash the display to try and avoid image retention
* Set the Black, White and Red stages at +15V and VCOM at Ground for steps A and C
* Pulse step A for 16 counts (I think this is milliseconds but I don't recall) followed by B, C and D for 4 counts
* Repeat that pulse pattern 3 more times

## Phase 1 - Clear to white, by pulling all inks down
* Set the Black, White and Red stages to -15v and VCOM at Ground for steps A and B
* Do... something... for the Red on step C and D. I'm not sure these are even effective
* Pulse step A for 16 counts, B C and D for 4, repeat 3 more times

## Phase 2 - Pull the black ink back toward the front
* Set the black stage to +15 for step B only and the white to -15 for step A only. VCOM at Ground.
* Pulse for 4, 8, 8, 16, repeated 15 more times. Note: In this case the last 8 and 16 might as well have been 0 and 24 since all elements are at Ground during those rest stages

## Phase 3, 4, 5 and 6

Unused since we're dealing with just black, but these would usually jiggle things around until we get some semblance of the sharpness/clarity/colour we want.

## Conclusion

From this we can conclude that the black particles will move toward the front of the display when pulsed with VSH1 (+15v) and will move away when pulse with VSL (-15v). If you're particularly brave you can see what happens when you set VCOM to -15v and black to +15v creating a 30v differential to give things a little extra kick. You'll notice I do this with the Yellow LUTS since that Yellow/Black mix is a pain to shunt around.

The long and short of it is that I don't know much about how *exactly* these LUTs should be set up to achieve a desired outcome and we do most of our LUTs via trial and error. Hopefully some of what I've detailed above will give you a better idea what you're telling the hardware to do!

dries007, 2019-06-29:
I hope no-one minds me posting this, but this video is the reason I ended up buying an Inky wHat.

["E-paper hacking: fastest possible refresh rate" By Applied Science on YouTube.](https://www.youtube.com/watch?v=MsbiO8EAsGw)

I think the display he has in the video is the same model as the Inky wHat, with potentially the same driver. In the video he explains in quite a bit of detail how e-ink displays work and some of the theory behind the LUTs. I think it's a valuable resource for anyone who wants to tinker with the displays beyond standard library usage.

I have seen the [DEPG0420R01V3.0.pdf](https://www.china-epaper.com/uploads/soft/DEPG0420R01V3.0.pdf) datasheet linked at various places, but only ever referring to the pHat. Does this one also match the wHat?