pimoroni/inky #21: smbus not found

TheRijn, 2019-01-31:
`smbus` wasn't installed automatically by the install script, it should include 
```sudo apt install python3-smbus``` and ```sudo apt install python2-smbus```.

Gadgetoid, 2019-02-05:
Should be fixed! Thanks for letting me know.

closed 2019-02-05 by Gadgetoid

fisherthewol, 2019-03-03:
This doesn't actually seem to be fixed as of 03/03/2019; installing the library to my venv did not also install any smbus library. 
I fixed it locally by `pip3 install smbus-cffi`, but that library hasn't been updated since 2015 (on pypi anyway). 
I'm more than happy to write a pull request replacing it with a drop in replacement (I'm thinking smbus2 https://pypi.org/project/smbus2/) and making sure it's a dependancy if that seems to be needed?

Gadgetoid, 2019-03-04:
Ah, this would be because your venv is either not using system packages, or using a version of Python for which the SMBus packages are not available, presumably?

The apt/pip mix of package sources has always been something of a thorn in my side and I've been contemplating moving to smbsu2 generally since I already use it in instances where I need raw i2c and not SMBus communication.

The switch suggested in #24 should work. I'll look into it.

fisherthewol, 2019-03-04:
Afaik using a venv means that I competely ignore system-wide packages, yes. The rpi I'm using is running python 3.5.(2?), Which does not appear to have a package just called SMBus on it's own.

I'll spend the afternoon testing the changes in #24 locally to see if it does work and get back to you.

Gadgetoid, 2019-03-04:
I've run into similar problems with my local test environments for non-standard versions of Python- there's simply no smbus package (unless I compile from source) since the C library file is named in the form `smbus.cpython-XXm-arm-linux-gnueabihf.so` for 3+ and `smbus.arm-linux-gbnueabihf.so` for 2.x.

And there's no `pip` version of smbus- although now we have arm wheel support it might be sensible to look into providing one.