pimoroni/inky #28: Issue running quotes-what.py

brentru, 2019-03-12:
Hi,

I'm not able to run the quotes-what.py demo on my Raspbery Pi. 

Running a fresh install raspbian latest, python 3, pip 3.

Issue is with installing `wikiquotes`, doesn't seem libxslt/libxml2 is installed by default and I can't find a suitable package that works without it.

Output from Pi:

```
  Running setup.py install for lxml ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-r071llkk/lxml/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-wex8tsge-record/install-record.txt --single-version-externally-managed --compile:
    Building lxml version 4.3.2.
    Building without Cython.
    ERROR: b'/bin/sh: 1: xslt-config: not found\n'
    ** make sure the development packages of libxml2 and libxslt are installed **

    Using build configuration of libxslt
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-3.5
    creating build/lib.linux-armv7l-3.5/lxml
    copying src/lxml/sax.py -> build/lib.linux-armv7l-3.5/lxml
    copying src/lxml/builder.py -> build/lib.linux-armv7l-3.5/lxml
    copying src/lxml/pyclasslookup.py -> build/lib.linux-armv7l-3.5/lxml
    copying src/lxml/_elementpath.py -> build/lib.linux-armv7l-3.5/lxml
    copying src/lxml/__init__.py -> build/lib.linux-armv7l-3.5/lxml
    copying src/lxml/cssselect.py -> build/lib.linux-armv7l-3.5/lxml
    copying src/lxml/ElementInclude.py -> build/lib.linux-armv7l-3.5/lxml
    copying src/lxml/doctestcompare.py -> build/lib.linux-armv7l-3.5/lxml
    copying src/lxml/usedoctest.py -> build/lib.linux-armv7l-3.5/lxml
    creating build/lib.linux-armv7l-3.5/lxml/includes
    copying src/lxml/includes/__init__.py -> build/lib.linux-armv7l-3.5/lxml/includes
    creating build/lib.linux-armv7l-3.5/lxml/html
    copying src/lxml/html/_setmixin.py -> build/lib.linux-armv7l-3.5/lxml/html
    copying src/lxml/html/_html5builder.py -> build/lib.linux-armv7l-3.5/lxml/html
    copying src/lxml/html/builder.py -> build/lib.linux-armv7l-3.5/lxml/html
    copying src/lxml/html/_diffcommand.py -> build/lib.linux-armv7l-3.5/lxml/html
    copying src/lxml/html/diff.py -> build/lib.linux-armv7l-3.5/lxml/html
    copying src/lxml/html/clean.py -> build/lib.linux-armv7l-3.5/lxml/html
    copying src/lxml/html/defs.py -> build/lib.linux-armv7l-3.5/lxml/html
    copying src/lxml/html/__init__.py -> build/lib.linux-armv7l-3.5/lxml/html
    copying src/lxml/html/ElementSoup.py -> build/lib.linux-armv7l-3.5/lxml/html
    copying src/lxml/html/soupparser.py -> build/lib.linux-armv7l-3.5/lxml/html
    copying src/lxml/html/html5parser.py -> build/lib.linux-armv7l-3.5/lxml/html
    copying src/lxml/html/usedoctest.py -> build/lib.linux-armv7l-3.5/lxml/html
    copying src/lxml/html/formfill.py -> build/lib.linux-armv7l-3.5/lxml/html
    creating build/lib.linux-armv7l-3.5/lxml/isoschematron
    copying src/lxml/isoschematron/__init__.py -> build/lib.linux-armv7l-3.5/lxml/isoschematron
    copying src/lxml/etree.h -> build/lib.linux-armv7l-3.5/lxml
    copying src/lxml/etree_api.h -> build/lib.linux-armv7l-3.5/lxml
    copying src/lxml/lxml.etree.h -> build/lib.linux-armv7l-3.5/lxml
    copying src/lxml/lxml.etree_api.h -> build/lib.linux-armv7l-3.5/lxml
    copying src/lxml/includes/xmlerror.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
    copying src/lxml/includes/relaxng.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
    copying src/lxml/includes/xpath.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
    copying src/lxml/includes/config.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
    copying src/lxml/includes/xmlparser.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
    copying src/lxml/includes/tree.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
    copying src/lxml/includes/etreepublic.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
    copying src/lxml/includes/dtdvalid.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
    copying src/lxml/includes/xmlschema.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
    copying src/lxml/includes/xinclude.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
    copying src/lxml/includes/htmlparser.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
    copying src/lxml/includes/schematron.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
    copying src/lxml/includes/xslt.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
    copying src/lxml/includes/uri.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
    copying src/lxml/includes/__init__.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
    copying src/lxml/includes/c14n.pxd -> build/lib.linux-armv7l-3.5/lxml/includes
    copying src/lxml/includes/lxml-version.h -> build/lib.linux-armv7l-3.5/lxml/includes
    copying src/lxml/includes/etree_defs.h -> build/lib.linux-armv7l-3.5/lxml/includes
    creating build/lib.linux-armv7l-3.5/lxml/isoschematron/resources
    creating build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/rng
    copying src/lxml/isoschematron/resources/rng/iso-schematron.rng -> build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/rng
    creating build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/xsl
    copying src/lxml/isoschematron/resources/xsl/RNG2Schtrn.xsl -> build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/xsl
    copying src/lxml/isoschematron/resources/xsl/XSD2Schtrn.xsl -> build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/xsl
    creating build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
    copying src/lxml/isoschematron/resources/xsl/iso-schematron-xslt1/iso_dsdl_include.xsl -> build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
    copying src/lxml/isoschematron/resources/xsl/iso-schematron-xslt1/iso_schematron_skeleton_for_xslt1.xsl -> build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
    copying src/lxml/isoschematron/resources/xsl/iso-schematron-xslt1/iso_schematron_message.xsl -> build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
    copying src/lxml/isoschematron/resources/xsl/iso-schematron-xslt1/iso_abstract_expand.xsl -> build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
    copying src/lxml/isoschematron/resources/xsl/iso-schematron-xslt1/iso_svrl_for_xslt1.xsl -> build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
    copying src/lxml/isoschematron/resources/xsl/iso-schematron-xslt1/readme.txt -> build/lib.linux-armv7l-3.5/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
    running build_ext
    building 'lxml.etree' extension
    creating build/temp.linux-armv7l-3.5
    creating build/temp.linux-armv7l-3.5/src
    creating build/temp.linux-armv7l-3.5/src/lxml
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-6waWnr/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DCYTHON_CLINE_IN_TRACEBACK=0 -Isrc -Isrc/lxml/includes -I/usr/include/python3.5m -c src/lxml/etree.c -o build/temp.linux-armv7l-3.5/src/lxml/etree.o -w
    In file included from src/lxml/etree.c:687:0:
    src/lxml/includes/etree_defs.h:14:31: fatal error: libxml/xmlversion.h: No such file or directory
     #include "libxml/xmlversion.h"
                                   ^
    compilation terminated.
    Compile failed: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    cc -I/usr/include/libxml2 -c /tmp/xmlXPathInitjm8no0xj.c -o tmp/xmlXPathInitjm8no0xj.o
    /tmp/xmlXPathInitjm8no0xj.c:1:26: fatal error: libxml/xpath.h: No such file or directory
     #include "libxml/xpath.h"
                              ^
    compilation terminated.
    *********************************************************************************
    Could not find function xmlCheckVersion in library libxml2. Is libxml2 installed?
    *********************************************************************************
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

    ----------------------------------------

```

Gadgetoid, 2019-03-13:
Try: `sudo apt install python3-lxml` for a precompiled version.

brentru, 2019-03-18:
@Gadgetoid  Looks like it works, closing...

closed 2019-03-18 by brentru