pimoroni/inky #31: FileNotFoundError: [Errno 2] No such file or directory

GarthDB, 2019-04-26:
Sorry, the title of the issue isn't very helpful, but I'm not sure what the error actually is.

I got through all the issues with Pil, and wikiquotes to get to the point of testing it using the `quotes-what.py` example, but I haven't found a solution to this error:

```shell
$ python3 quotes-what.py --colour red
Inky wHAT: Quotes

Display quotes on Inky wHAT.

"Science and everyday life cannot
  and should not be separated."
- Rosalind Franklin

Traceback (most recent call last):
  File "quotes-what.py", line 165, in <module>
    inky_display.show()
  File "/usr/local/lib/python3.5/dist-packages/inky/inky.py", line 288, in show
    self._update(buf_a, buf_b)
  File "/usr/local/lib/python3.5/dist-packages/inky/inky.py", line 209, in _update
    self.setup()
  File "/usr/local/lib/python3.5/dist-packages/inky/inky.py", line 184, in setup
    self._spi.open(0, self.cs_pin)
FileNotFoundError: [Errno 2] No such file or directory
```

For context, I'm on a raspberry pi zero w with the latest Raspbian Stretch Lite (Version: April 2019).

I'd appreciate any troubleshooting recommendations.

GarthDB, 2019-04-26:
I think it might be a connection issue - I'll try a different raspberry pi tomorrow.

Gadgetoid, 2019-04-26:
It looks like you need to enable SPI. Run `sudo raspi-config` and you'll find it in Interfacing Options.

GarthDB, 2019-04-27:
That was it @Gadgetoid - Thanks!

I can make a pr to update the readme with that info if it would be helpful to future newbs like me.

https://twitter.com/GarthDB/status/1122200968852987905

GarthDB, 2019-04-27:
Or I guess I could add a check to the example to make sure the SPI has been enabled or give a decent error message.

closed 2019-04-27 by GarthDB