pimoroni/inky #33: weather-phat.py ImportError: No module named font_fredoka_one

zicmama, 2019-06-06:
pi@raspberrypi:~/inky/examples/phat $ ./weather-phat.py -c yellow
Traceback (most recent call last):
  File "./weather-phat.py", line 9, in <module>
    from font_fredoka_one import FredokaOne
ImportError: No module named font_fredoka_one

Font error also occurs on name-badge.py

zicmama, 2019-06-06:
sudo pip install font_fredoka_one
sudo pip install font_hanken_grotesk

zicmama, 2019-06-06:
```
pi@raspberrypi:~/inky/examples/phat $ ./weather-phat.py -c yellow
Inky pHAT: Weather

Displays weather information for a given location. The default location is Sheffield-on-Sea.

Traceback (most recent call last):
  File "./weather-phat.py", line 99, in <module>
    weather = get_weather(location_string)
  File "./weather-phat.py", line 63, in get_weather
    soup = BeautifulSoup(res.content, "lxml")
  File "/usr/local/lib/python2.7/dist-packages/bs4/__init__.py", line 196, in __init__
    % ",".join(features))
bs4.FeatureNotFound: Couldn't find a tree builder with the features you requested: lxml. Do you need to install a parser library?
```

zicmama, 2019-06-06:
sudo pip install lxml

zicmama, 2019-06-06:
```
Collecting lxml
  Downloading https://files.pythonhosted.org/packages/7d/29/174d70f303016c58bd790c6c86e6e86a9d18239fac314d55a9b7be501943/lxml-4.3.3.tar.gz (4.4MB)
    100% |████████████████████████████████| 4.4MB 18kB/s 
Building wheels for collected packages: lxml
  Running setup.py bdist_wheel for lxml ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-Yirncw/lxml/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpULTRyUpip-wheel- --python-tag cp27:
  Building lxml version 4.3.3.
  Building without Cython.
  ERROR: /bin/sh: 1: xslt-config: not found
  
  ** make sure the development packages of libxml2 and libxslt are installed **
  
  Using build configuration of libxslt
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv6l-2.7
  creating build/lib.linux-armv6l-2.7/lxml
  copying src/lxml/usedoctest.py -> build/lib.linux-armv6l-2.7/lxml
  copying src/lxml/sax.py -> build/lib.linux-armv6l-2.7/lxml
  copying src/lxml/ElementInclude.py -> build/lib.linux-armv6l-2.7/lxml
  copying src/lxml/cssselect.py -> build/lib.linux-armv6l-2.7/lxml
  copying src/lxml/__init__.py -> build/lib.linux-armv6l-2.7/lxml
  copying src/lxml/pyclasslookup.py -> build/lib.linux-armv6l-2.7/lxml
  copying src/lxml/_elementpath.py -> build/lib.linux-armv6l-2.7/lxml
  copying src/lxml/builder.py -> build/lib.linux-armv6l-2.7/lxml
  copying src/lxml/doctestcompare.py -> build/lib.linux-armv6l-2.7/lxml
  creating build/lib.linux-armv6l-2.7/lxml/includes
  copying src/lxml/includes/__init__.py -> build/lib.linux-armv6l-2.7/lxml/includes
  creating build/lib.linux-armv6l-2.7/lxml/html
  copying src/lxml/html/usedoctest.py -> build/lib.linux-armv6l-2.7/lxml/html
  copying src/lxml/html/_html5builder.py -> build/lib.linux-armv6l-2.7/lxml/html
  copying src/lxml/html/_setmixin.py -> build/lib.linux-armv6l-2.7/lxml/html
  copying src/lxml/html/html5parser.py -> build/lib.linux-armv6l-2.7/lxml/html
  copying src/lxml/html/soupparser.py -> build/lib.linux-armv6l-2.7/lxml/html
  copying src/lxml/html/defs.py -> build/lib.linux-armv6l-2.7/lxml/html
  copying src/lxml/html/clean.py -> build/lib.linux-armv6l-2.7/lxml/html
  copying src/lxml/html/__init__.py -> build/lib.linux-armv6l-2.7/lxml/html
  copying src/lxml/html/ElementSoup.py -> build/lib.linux-armv6l-2.7/lxml/html
  copying src/lxml/html/builder.py -> build/lib.linux-armv6l-2.7/lxml/html
  copying src/lxml/html/_diffcommand.py -> build/lib.linux-armv6l-2.7/lxml/html
  copying src/lxml/html/diff.py -> build/lib.linux-armv6l-2.7/lxml/html
  copying src/lxml/html/formfill.py -> build/lib.linux-armv6l-2.7/lxml/html
  creating build/lib.linux-armv6l-2.7/lxml/isoschematron
  copying src/lxml/isoschematron/__init__.py -> build/lib.linux-armv6l-2.7/lxml/isoschematron
  copying src/lxml/etree.h -> build/lib.linux-armv6l-2.7/lxml
  copying src/lxml/etree_api.h -> build/lib.linux-armv6l-2.7/lxml
  copying src/lxml/lxml.etree.h -> build/lib.linux-armv6l-2.7/lxml
  copying src/lxml/lxml.etree_api.h -> build/lib.linux-armv6l-2.7/lxml
  copying src/lxml/includes/uri.pxd -> build/lib.linux-armv6l-2.7/lxml/includes
  copying src/lxml/includes/schematron.pxd -> build/lib.linux-armv6l-2.7/lxml/includes
  copying src/lxml/includes/xpath.pxd -> build/lib.linux-armv6l-2.7/lxml/includes
  copying src/lxml/includes/xmlerror.pxd -> build/lib.linux-armv6l-2.7/lxml/includes
  copying src/lxml/includes/xmlparser.pxd -> build/lib.linux-armv6l-2.7/lxml/includes
  copying src/lxml/includes/xinclude.pxd -> build/lib.linux-armv6l-2.7/lxml/includes
  copying src/lxml/includes/relaxng.pxd -> build/lib.linux-armv6l-2.7/lxml/includes
  copying src/lxml/includes/__init__.pxd -> build/lib.linux-armv6l-2.7/lxml/includes
  copying src/lxml/includes/xmlschema.pxd -> build/lib.linux-armv6l-2.7/lxml/includes
  copying src/lxml/includes/etreepublic.pxd -> build/lib.linux-armv6l-2.7/lxml/includes
  copying src/lxml/includes/htmlparser.pxd -> build/lib.linux-armv6l-2.7/lxml/includes
  copying src/lxml/includes/tree.pxd -> build/lib.linux-armv6l-2.7/lxml/includes
  copying src/lxml/includes/config.pxd -> build/lib.linux-armv6l-2.7/lxml/includes
  copying src/lxml/includes/c14n.pxd -> build/lib.linux-armv6l-2.7/lxml/includes
  copying src/lxml/includes/xslt.pxd -> build/lib.linux-armv6l-2.7/lxml/includes
  copying src/lxml/includes/dtdvalid.pxd -> build/lib.linux-armv6l-2.7/lxml/includes
  copying src/lxml/includes/etree_defs.h -> build/lib.linux-armv6l-2.7/lxml/includes
  copying src/lxml/includes/lxml-version.h -> build/lib.linux-armv6l-2.7/lxml/includes
  creating build/lib.linux-armv6l-2.7/lxml/isoschematron/resources
[... 118 lines cut ...]
    building 'lxml.etree' extension
    creating build/temp.linux-armv6l-2.7
    creating build/temp.linux-armv6l-2.7/src
    creating build/temp.linux-armv6l-2.7/src/lxml
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-Ub3vap/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DCYTHON_CLINE_IN_TRACEBACK=0 -Isrc -Isrc/lxml/includes -I/usr/include/python2.7 -c src/lxml/etree.c -o build/temp.linux-armv6l-2.7/src/lxml/etree.o -w
    In file included from src/lxml/etree.c:687:0:
[... cut ...]
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-Yirncw/lxml/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-Sn7Z5e-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-Yirncw/lxml/

```

Gadgetoid, 2019-06-12:
In case where pip packages wont build, you can try ones from apt. In this case either:

```sudo apt install python-lxml```

or

```sudo apt install python3-lxml```

zicmama, 2019-06-12:
worked for me! Thanks

closed 2019-06-12 by zicmama