pimoroni/inky #34: [Question/Performance] Only convert and send bits for colour screen when red or yellow

nyaray, 2019-06-08:
While reading the source for inspiration for an elixir port I'm contributing to, I found that there seems to be an unnecessary conversion of the pixels in the accent colour (currently, red or yellow) to bits, even though the screen might be black.

_My question_: Is it necessary to convert and transmit bits for red/yellow on black screens due to a quirk in the chip(s)/the implementation or can it be skipped?

The code in question: https://github.com/pimoroni/inky/blob/master/library/inky/inky.py#L278-L288

Gadgetoid, 2019-06-12:
As long as your LUTs are configured to create no voltage difference across the screen for Black/Red then you can - I believe - safely ignore this memory portion. This would give you a true black/white screen with no possibility of an interim shade.

IIRC our default Black/White configuration actually uses these bits to provide a shade of grey using the existing black ink particles.

To avoid affecting the screen from any uninitialised data that might be in memory at power on, you should ensure that the values for VCOM and Yellow/Red remain the same.

It may also be sufficient to write this data only once at power-on and omit it for subsequent display updates.

So, TLDR: It's not necessary, but you've got to set the values or LUTs so you don't accidentally refresh garbage data to your screen.

nyaray, 2019-06-19:
> As long as your LUTs are configured to create no voltage difference across the screen for Black/Red then you can - I believe - safely ignore this memory portion.

Ah, noted.

> IIRC our default Black/White configuration actually uses these bits to provide a shade of grey using the existing black ink particles.

On board or in software? Cool trick, either way!

I guess we'll just leave it as-is then, and have an open issue for when/if someone actually gets a b/w one and cares about performance (heh).

lawik, 2019-07-26:
@Gadgetoid Would the shade of gray only work for the black/white devices or would it work for running a black/white/red device in black/white mode? Would gray just be the way red pixels would render then?

closed 2023-01-20 by Gadgetoid