pimoroni/inky #35: Cannot import from inky

bjoernbusch, 2019-06-30:
I have a Raspberry Pi Zero with a clean fresh install of the Raspbian image from April, so still with Stretch. I did nothing but update the locale to en-us and set the timezone. Then I install inky as described in the readme.md and it is successful. I try to create a script with only
```
from inky import InkyPHAT
inkyphat = InkyPHAT('red')
inkyphat.set_border(inky.RED)
inkyphat.show()
```
but when I try to run it I get the following error:

Traceback (most recent call last):
  File "inky.py", line 1, in <module>
    from inky import InkyPHAT
  File "/home/pi/inky/inky.py", line 1, in <module>
    from inky import InkyPHAT
ImportError: cannot import name InkyPHAT

I tried the same with the new Buster image, same issue. I updated both to the latest packages and kernel, same error.

I'd appreciate any hint on what stupid little thing I'm not seeing.

bjoernbusch, 2019-06-30:
ok, so using the script from the pimoroni intro explanation brings everything to life.

https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-inky-phat

Gadgetoid, 2019-07-01:
What happens if you:

```
import inky
dir(inky)
````

There was a previous package by the name Inky and I wonder if that's shadowing your install somehow.