pimoroni/inky #36: Example script fonts not installed in Raspbian Buster

jjsanderson, 2019-07-17:
Though #2 appears to have resolved this some time ago, a fresh install on Raspbian Buster resulted in font errors when running the name-badge.py example script.

### Steps to reproduce

1. Clean install of Raspbian 2019-07-10 release
2. Install inky phat library via pip
3. run name-badge.py script

### Steps to resolve

Install missing fonts via pip.

Gadgetoid, 2019-07-23:
Since they are dependencies of the examples and not the library itself, these dependencies are only installed if you run `curl -sSL https://get.pimoroni.com/inky | bash`

I wonder if there's a better way to go about and/or document this- I guess the example itself could use a header that documents its dependencies.

But then, is Inky even useful without fonts? Maybe they *should* be dependencies of the library.

jjsanderson, 2019-07-23:
Hmm. I see the dilemma.

Wrapping the examples' font imports with `try` and a helpful error message tends to be my preferred way of doing this; it seems the most helpful approach for newcomers. However, it's messy for those who know what they're doing (I'm more commonly in the former camp than the latter, I think).

Perhaps the best compromise in this case would be a section in the repo readme? I guess the objective is to help people get up and running as smoothly as possible, in which case directing them to the examples (and their dependencies) will get them to the point of believing their pHAT works, quickly.

closed 2023-01-20 by Gadgetoid