pimoroni/inky #37: Use inky via pi zero GPIO expander mode

gigawatts, 2019-08-03:
Feature Request

I'm interested in using this library with the GPIO expander mode from a PC (https://www.raspberrypi.org/magpi/pi-zero-gpio-expander/). That way I could plug in the pi 0 with an inky phat display, update the inky display, and unplug. No OS required.

How hard would this be to port and use in that fashion?

Gadgetoid, 2019-08-20:
In my experience it's best to port the underlying transport libraries by "shimming" some custom code in their place. I'm surprised there's not a general purpose solution to this problem already out there.

You would need to create fake versions of `spidev` and `RPi.GPIO` that you install locally on your PC (or just have in the same folder, or virtual env, as your code). These fake versions would implement all the method calls that Inky uses (not many) and redirect them to equivalent GPIO Zero calls:

```
open(bus, cs)
max_speed_hz (property)
xfer3 (or xfer)
```

And do the same for RPi.GPIO calls:

```
setmode
setwarnings
setup
output
input
```

I can't say how fast this might be, but it should work.

Gadgetoid, 2019-08-27:
Some of the changes proposed to help support Inky pHAT simulation should also help you here.

These include the ability to specify an SPI bus, I2C bus and GPIO object to replace `spidev`, `smbus2` and `RPi.GPIO`. ie:

```python
inky = InkyPHAT(spi_bus=my_spi_bus, i2c_bus=my_i2c_bus, gpio=my_gpio)
```

You will, however, still need to write some code to translate the method calls.