pimoroni/inky #38: Various documentation fixes

shukriadams, 2019-08-13:
I just got my Inky, and it took a bit of investigating to get it figured out. The  documentation provided here is missing some important details, would be great if you could add them. Note, I was using Buster, I'm not sure of these errors aren't valid for previous versions.

1. Inky requires Python 3, it does not work on 2.
2. You need to install a whole bunch of external dependencies with

    curl https://get.pimoroni.com/inky | bash

this is listed on https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-inky-phat, it would be nice to have them here too.
3. The code examples in the readme of this repo and on the above Pimoroni don't work - I get lots of errors of type

        OSError: [Errno 2] No such file or directory: '/dev/i2c-1'

4. To display a custom image on your Inky, it _must_ be in Indexed color mode, and your color table must be locked to the colors of your Inky (I have a red unit, so my allowed colors were 000000, ffffff, and ff0000)

Gadgetoid, 2019-08-20:
1. What errors did you encounter with Python 2? It *should* work.

2. Slowly working on ways to make this better. It bugs me that our magical installer scripts are not installed in the repositories that they belong to and I've started taking steps to fix that.

3. That error usually occurs if you've not enabled i2c - which the install script should handle. What do you see if you `ls /dev/i2c-*`?

4. You can use any three colours since Inky only cares about the indexes (always treating them as black, white and red in order) but you'd probably want to use the ones you mentioned for consistency!

Anyway, some good points - thanks! We're always keen to improve documentation.

shukriadams, 2019-08-21:
Will get back after I burn a sandbox Raspbian SD so I can get some more meaningful error messages for you.

shukriadams, 2019-08-21:
I did a fresh install of Buster mini, out of the box, the default python --version returns 2.7.16.

When I run

    sudo curl https://get.pimoroni.com/inky | bash

I get lot of log output, most of which I've shorted below for brevity, but its clearly trying to install several modules which are named for python 3. It also tries to install several modules, fails, tries to fallback to pip defaults, and seems to fail there too

    Installing python3-pip...
    python3-numpy is required
    python3-pil is required
    python3-lxml is required
    python3-smbus is required
    Installing Inky pHAT & wHAT library for Python 3...

    Installing python-geocoder...
    E: Unable to locate package python-geocoder
    Apt failed to install python-geocoder!
    Falling back on pypi...

    Installing python3-geocoder...
    E: Unable to locate package python3-geocoder
    python3-beautifulsoup4 is required
    Installing python3-beautifulsoup4...
    E: Unable to locate package python3-beautifulsoup4
    Apt failed to install python3-beautifulsoup4!
    Falling back on pypi...
    Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple

    python3-icalendar is required
    Installing python3-icalendar...
    python-fonts is required
    Installing python-fonts...
    E: Unable to locate package python-fonts
    Apt failed to install python-fonts!

    python3-fonts is required
    Installing python3-fonts...
    E: Unable to locate package python3-fonts
    Apt failed to install python3-fonts!
    Falling back on pypi...
    python3-font_intuitive is required
    python3-font_fredoka_one is required
    python3-font_source_serif_pro is required
    Installing python3-font_source_sans_pro...
    python3-wikiquotes is required

    Some packages could not be installed, review the output for details!

Finally, when I run

    pi@raspberrypi:~ $ cd ~/Pimoroni/inky/examples
    pi@raspberrypi:~/Pimoroni/inky/examples $ python logo.py -t what -c red

I get the following error

    Inky pHAT/wHAT: Logo
    Displays the Inky pHAT/wHAT logo.
    Traceback (most recent call last):
      File "logo.py", line 28, in <module>
        inky_display = InkyWHAT(colour)
      File "/usr/local/lib/python2.7/dist-packages/inky/what.py", line 27, in __init__
        v_flip=False)
      File "/usr/local/lib/python2.7/dist-packages/inky/inky.py", line 74, in __init__
        self.eeprom = eeprom.read_eeprom()
      File "/usr/local/lib/python2.7/dist-packages/inky/eeprom.py", line 103, in read_eeprom
        i2c = SMBus(1)
      File "/usr/local/lib/python2.7/dist-packages/smbus2/smbus2.py", line 253, in __init__
        self.open(bus)
      File "/usr/local/lib/python2.7/dist-packages/smbus2/smbus2.py", line 265, in open
        self.fd = os.open("/dev/i2c-{}".format(bus), os.O_RDWR)
    OSError: [Errno 2] No such file or directory: '/dev/i2c-1'

when I do 

    ls /dev/i2c-*

I get no such file or directory. The above works perfectly if I force my user python version to 3.7, and from the module names I assumed version 3.x was probably a requirement.

shukriadams, 2019-08-21:
For images, if I try to render an indexed image with more than 3 images in the index, or something that is not indexed at all, I get

    Traceback (most recent call last):
      File "image.py", line 45, in <module>
        inky_display.set_image(img)
      File "/usr/local/lib/python3.7/dist-packages/inky/inky.py", line 335, in set_image
        self.buf = numpy.array(image, dtype=numpy.uint8).reshape((self.width, self.height))
    ValueError: cannot reshape array of size 366276 into shape (400,300)

shukriadams, 2019-08-21:
re : indexed colors, it seems we were both off? black-white-red causes the image to look inverted. white-black-red renders properly.

Gadgetoid, 2019-08-27:
You'll need to enable i2c on your Pi, which adds the `/dev/i2c-1` device.

This *should* be handled by our `curl get.pimoroni.com/inky | bash` installer, but it seems like it's bailing out with the other errors and possibly not managing that step.

TLDR: use `raspi-config` or the Pi Menu -> Preferences -> Raspberry Pi Configuration, then navigate to "Interfaces" and make sure you have "i2c" enabled.

These `curl | bash` scripts are a little bit of a legacy thorn in my side at the moment- I'm planning to move these all to better, more focussed install scripts within each repository.

shukriadams, 2019-09-04:
Think we can close this ticket, as it's more of a discussion than an actual bug report. Going Python 3 is an acceptable workaround for me.

closed 2019-09-04 by shukriadams