pimoroni/inky #41: inkyphat doesn't react at all

vienoz, 2019-09-24:
I did everything like in the "getting started" guide and when I use the examples I get expected outputs like " Hello... my name is:" and the description what is does. However on the device itself nothing happens, I only see the image that it arrived with. I have no Idea where to start looking for the problem

vienoz, 2019-10-01:
Well...yes. I got a glass of water, ran the code without changing a thing an it worked. No Idea how no idea why I didn't touch anything it just worked. However even now after booting, the first view attemps at displaying something dont't work. Sorry i can't be of more help, maybe have a look in Pimoroni's tech support forum.

sidelux, 2019-10-01:
Thanks anyway :')