pimoroni/inky #42: Setting border to red / yellow - is it possible?

gauntface, 2019-09-25:
I tried setting the border to red in one of the examples and found that it didn't work, only black and white work - is that expected?

Gadgetoid, 2019-09-27:
As far as I'm aware it should be possible to set the border to Red on all Red/Black/White displays and Yellow on all Yellow/Black/White displays. This is explicitly supported in the library, but it's possible that there's a bug preventing it from working properly. This warrants some investigation.

What version/type of Inky pHAT do you have?

gauntface, 2019-09-27:
I have an inky what red, not sure what version - how can I tell?

dustinrb, 2020-10-09:
I'm having the same issue. I'm using a black/yellow SSD1608 variant. Looking through the source code I see that the border color is set in the Inky class, but the WRITE_BORDER command is never given to the board in the _update method.

jerbzz, 2021-03-01:
@dustinrb https://github.com/pimoroni/inky/issues/100#issuecomment-787552450

closed 2023-01-20 by Gadgetoid