pimoroni/inky #43: Cannot install Python libraries for pHat

C-D-Lewis, 2019-10-04:
I'm trying to install the python library for my brank new Inky pHat, but I get the following error with both the following commands:
`curl https://get.pimoroni.com/inky | bash`
`sudo pip install inky`

Any ideas? My Pi version:
```
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 8.0 (jessie)
Release:	8.0
Codename:	jessie
```

Gadgetoid, 2019-10-05:
I'm not sure what's happened here. The issue appears to be with the `smbus2` package, but glancing over the source/GitHub history suggests it hasn't changed meaningfully in a while.

It might be worth trying: `sudo pip install smbus2=0.2.3` followed by `sudo pip install inky` to see if it works around the problem.

Gadgetoid, 2019-10-05:
I grabbed the latest smbus2 package to inspect, and `extras_require` looks fine. I suspect your downloaded/extracted file was corrupted, or some other temporary issue occured. You could also try:

```
sudo pip --no-cache-dir install smbus2
```

C-D-Lewis, 2019-10-05:
Thanks @Gadgetoid for the swift feedback. I tried the first command to little success:
```
pi@raspberrypi:~ $ sudo pip install smbus2=0.2.3
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 269, in run
    InstallRequirement.from_line(name, None))
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 173, in from_line
    return cls(req, comes_from, url=url, prereleases=prereleases)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 71, in __init__
    req = pkg_resources.Requirement.parse(req)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2793, in parse
    reqs = list(parse_requirements(s))
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2721, in parse_requirements
    "version spec")
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2686, in scan_list
    raise ValueError(msg, line, "at", line[p:])
ValueError: ('Expected version spec in', 'smbus2=0.2.3', 'at', '=0.2.3')
```

Also

```
pi@raspberrypi:~ $ sudo pip --no-cache-dir install smbus2

Usage:
  pip <command> [options]

no such option: --no-cache-dir
```

Perhaps I need a new Python version or a cleaner environment?

Gadgetoid, 2019-10-07:
Oof I missed an `=` in my command, despite trying not to miss an `=` in my command. Try:

```sudo pip install smbus2==0.2.3```

The lack of a `--no-cache-dir` option might suggest a very out of date smbus, worth trying a `sudo apt update` followed by `sudo apt upgrade` and then `sudo pip install --upgrade pip`.

Is this an old Pi setup on Raspbian Jessie?

C-D-Lewis, 2019-10-10:
Thanks @Gadgetoid, I will re-attempt this weekend!

C-D-Lewis, 2019-10-12:
@Gadgetoid Still little success, sadly. Here is the versions I'm using:
Hardware: Pi Zero W
Software: Raspbian GNU/Linux 8 (jessie)
Python: 2.7.9 (`/usr/bin/python`)
pip: pip 19.2.3 (`/usr/local/bin/pip`)

And yet the same problem with installing smbus persists. Is it worth trying a completely fresh image?

C-D-Lewis, 2019-10-12:
After a fresh Buster install and over an hour and a half of compiling `numpy`, it finished installing successfully! However, I now have another issue (sadly!)

```
Python 2.7.16 (default, Apr  6 2019, 01:42:57)
[GCC 8.2.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import inky
>>> from inky import InkyPHAT
>>> inkyphat = InkyPHAT('yellow')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/inky/phat.py", line 27, in __init__
    v_flip=False)
  File "/usr/local/lib/python2.7/dist-packages/inky/inky.py", line 74, in __init__
    self.eeprom = eeprom.read_eeprom()
  File "/usr/local/lib/python2.7/dist-packages/inky/eeprom.py", line 103, in read_eeprom
    i2c = SMBus(1)
  File "/usr/local/lib/python2.7/dist-packages/smbus2/smbus2.py", line 279, in __init__
    self.open(bus)
  File "/usr/local/lib/python2.7/dist-packages/smbus2/smbus2.py", line 308, in open
    self.fd = os.open(filepath, os.O_RDWR)
OSError: [Errno 2] No such file or directory: '/dev/i2c-1'
```

I did some Googling and determined that i2c was not enabled. I followed [these instructions](https://www.runeaudio.com/forum/how-to-enable-i2c-t1287.html#) (with the exception of using `apt` instead of `pacman` to install the I2C tools, and after a reboot I could use `InkyPHAT('yellow')` without the same error. 

However, when trying to follow the 'set border' example in the `README.md` file, (which should read `inkyphat.BLACK` etc instead of `ink.BLACK`, I got this:
```
>>> inkyphat.set_border(inkyphat.BLACK)
>>> inkyphat.show()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/inky/inky.py", line 325, in show
    self._update(buf_a, buf_b, busy_wait=busy_wait)
  File "/usr/local/lib/python2.7/dist-packages/inky/inky.py", line 238, in _update
    self.setup()
  File "/usr/local/lib/python2.7/dist-packages/inky/inky.py", line 213, in setup
    self._spi.open(0, self.cs_pin)
IOError: [Errno 2] No such file or directory
```
After more Googling and using `raspi-config` to enable the SPI interface, I am now able to use the display! 😄 

Closing this now.

closed 2019-10-12 by C-D-Lewis