pimoroni/inky #48: Required GPIO pins for Inky pHAT

JimMadge, 2020-03-12:
Is there a definitive list of the pins required to use the Inky pHAT?

The inky module lists the following pins,

```Python
RESET_PIN = 27
BUSY_PIN = 17
DC_PIN = 22

MOSI_PIN = 10
SCLK_PIN = 11
CS0_PIN = 0
```
while pinout.xyz indicates the CS pin [should be BCM8](https://pinout.xyz/pinout/inky_phat). The `MOSI_PIN` and `SCLK_PIN` variables also appear to be unused.

When I connected only the pins from the Python module (along with 3.3V and 5V power). Writing to the display hangs in `_busy_wait`. When I connect the pins from the pinout.xyz page, there is no hang but the display is not changed.

Making a mistake in connecting the jumpers notwithstanding, are there more pins that need to be connected? Is there something I have missed?

Gadgetoid, 2020-03-12:
Chip-Select 0 is, indeed, BCM8. That `CS0_PIN` variable should probably just be `CS0` since it refers to an SPI Chip-Select channel and *not* a BCM pin. MOSI and SCLK are also implicit to SPI port 0- so `spidev.SpiDev(0, 0)` uses BCM 10, BCM 9, BCM 11 and BCM 8.

You should also ensure all the Groudn connections are made and, potentially, also Physical Pin 17 (the other 3.3v power)

JimMadge, 2020-03-14:
Thanks for your help, here is what I found.

Chip pins,
- BCM 17 (Chip Busy)
- BCM 27 (Chip Reset)
- BCM 22 (Chip Command)

SPI pins,
- BCM 8 (SPI Chip Select)
- BCM 9 (SPI MISO)
- BCM 10 (SPI MOSI)
- BCM 11 (SPI MISO)

Power pins,
- Physical pin 1 (3v3)
- Physical pin 2 (5v)
- Any ground pin (except 30!?) (6, 9, 14, 20, 25, 34, 39)

I wonder if the ground pin issue is just my board. Would you like me to make a pull request documenting these pins (and possibly renaming the CS0_PIN variable) somewhere in the repository?

I will make a pull request on https://github.com/gadgetoid/Pinout.xyz to add the missing SPI pin.

Gadgetoid, 2020-03-16:
I don't believe MISO (BCM11) is phsyically connected, but otherwise this seems sensible.

Documentation is always welcome.. and- yes- that `CS0_PIN` variable could stand to have less confusing name.

JimMadge, 2020-03-17:
I had another chance to check. I was mistaken, MISO is not required.

PR #49 (2020-06-12): Update docs
Adding some clarification about the pins got a bit out of hand. I have,

- Added some notes about the pins to inky.py
- Renamed `CS0_PIN` and `cs_pin` to `CS0` and `cs_channel` respectively
- Exposed the colours at the modules top level, which was suggested in the README
- Updated the Sphinx documentation
  - Added type indications to arguments
  - Used intersphinx to link to external documentation where possible
  - Added links to the tutorials, repository and store pages

I hope this is OK/helpful.

closes #48
closed 2020-06-12 by Gadgetoid