pimoroni/inky #50: PIL library not installed by setup.py

cipy, 2020-04-18:
the PIL library is required by examples

also missing libraries required by examples: font_fredoka_one, font_hanken_grotesk, font_intuitive

Gadgetoid, 2020-04-20:
Adding PIL as a dependency on Raspberry Pi tends to lead to pip pulling down the source package and attempting to compile it- usually unsucessfully.

Example dependencies shouldn't generally by specified by setup.py because the examples aren't library code. We have an install.sh solution for this with a custom setup.cfg section, but it hasn't been backported to Inky yet.

Thinking about it- It also doesn't support additional requirements via pip, so maybe a requirements.txt in the examples folder is worth implementing.

cipy, 2020-04-20:
True, the examples are something else. Thanks 🙃

closed 2023-01-20 by Gadgetoid

cpietsch, 2024-12-15:
per default rapsberry pi zero + bookworm tries to compile Pillow when running the setup and fails

```
Building wheel for pillow (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for pillow (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [217 lines of output]

  The headers or library files could not be found for jpeg,
      a required dependency when compiling Pillow from source.

      Please see the install instructions at:
         https://pillow.readthedocs.io/en/latest/installation/basic-installation.html
```