pimoroni/inky #53: Inky pHAT revisions

DurandA, 2020-06-18:
When trying to merge [a port](https://github.com/DurandA/inky-cryptochart/pull/1) of my [inky-cryptochart project](https://github.com/DurandA/inky-cryptochart) to this newer _inky_ library from the older [_inky-phat_](https://github.com/pimoroni/inky-phat) library, I noticed that both my Inky pHAT and my [Waveshare E-Ink display HAT](https://github.com/pimoroni/inky-phat/issues/17) stopped working.

As stated in your project description, this library does not support the older red v1 revision. Is there a specific reason for it? Will I hit a wall if I try to port the driver to this library? I don't think these Inky pHATs are open hardware, but can you share the changes between the different revisions? Do they use a different display or driver chip?

Also, I noticed a [`feature/ssd1608`](https://github.com/pimoroni/inky/tree/feature/ssd1608) branch. Is this for a future revision?

Gadgetoid, 2020-06-18:
There are myriad different resolution displays, different driver ICs and different configuration profiles (LUTs) for each display. The primary motivation for dropping v1 support from this library was to keep it simple, and uncomplicated by having to support multiple drivers. If you look over the v1 code you'll notice a really clunky approach to supporting multiple displays that Inky was attempting to reconsile.

Of course... e-ink manufacturers don't seem to give us a break! Indeed the `feature/ssd1608` branch is intended to add support for a future revision. So that simplicity has somewhat melted away anyway.

There's nothing stopping the v1 code from being ported to this library using the same approach as the ssd1608, but the result probably wouldn't be very elegant.

We now populate an EEPROM on each Inky display in order to store information about the type and resolution of each display, but our current approach to reading and acting upon that EEPROM is suboptimal, too.

DurandA, 2020-06-18:
Thanks for the extensive answer, this is appreciated.

May I ask what is the reference of the display of the v1, v2 and v3 red version? If I understand correctly, v2 and v3 use a different display but the same driver whilst v1 uses another driver. Also, can both v2 and v3 be used in monochrome ("black") mode for fast display?

Gadgetoid, 2020-06-19:
I believe all the displays will do some kind of monochrome fast display, even the yellow/red displays can be driven slightly faster if you're willing to compromise on image quality. In all cases, though, they require a new LUT configuration and there's no documentation (that I'm aware of) to guide this process except the stuff I wrote after bashing my head against them for several weeks: https://github.com/pimoroni/inky/blob/feb2303e377badbf4cf940e8e5e309da6a2e5168/library/inky/inky.py#L108-L156

I'm afraid I've completely lost the plot in terms of which display driver/display part goes with which generation- I had a smattering of datasheets but no way to correlate them to a particular board. I juggle so many components and corresponding libraries. I can find out if engineering has records, though.

Sorry I can't be more immediately helpful.

closed 2023-01-20 by Gadgetoid