pimoroni/inky #55: Problems with the calendar-phat.py

kompas1, 2020-08-07:
So I have installed everything on a clean Buster-Installation, i could clean the screen but as soon as i want to use the calendar-phat.py i get the following output:

 _Inky pHAT: Calendar
Draws a calendar for the current month to your Inky pHAT.
This example uses a sprite sheet of numbers and month names which are
composited over the background in a couple of different ways.

Traceback (most recent call last):
  File "calendar-phat.py", line 6, in <module>
    import calendar
  File "/home/pi/Pimoroni/inky/examples/phat/calendar.py", line 85, in <module>
    text = Image.open(os.path.join(PATH, "resources/calendar.png"))
NameError: name 'Image' is not defined_

Does anyone knows why? This happened with every Installation i had. The other things also didn't worked. It actually bothers me a bit because I have no idea why it is happening

Gadgetoid, 2020-08-07:
It looks like you've invoked one of Python's more irritating quirks by creating a file called `calendar.py` alongside `calendar-phat.py`.

When `calendar-phat.py` tries to `import calendar` - a library for... doing calendar stuff 😆 instead it finds `calendar.py` alongside the Python file and tries to import that instead.

Basically- never name a Python file after a library you want to import, or might want to import in another Python file in the same directory and you'll be fine.

kompas1, 2020-08-08:
okay thanks, that actually solved the problem, makes sense actually, i don't know why i didn't came up with checking that on my own :man_facepalming: . At least the first one so now it's time for me to try the second part. :smile:

kompas1, 2020-08-08:
@Gadgetoid but another thing i noticed is, that I still can see some of the older things i put on the screen. Example from just now: I put on the weather-phat.py just to check it, worked easy. Now i put on the calendar-phat.py, also works fine but i can still see some data from the weather-phat.py in the 'background'. Will it stay like that or does it just take some time to adjust? Because I also wiped the whole screen with the clean.py but there is still some 'background' visible

kompas1, 2020-08-08:
So heres my try of removing the backdrop.png from the calendar file:

#!/usr/bin/env python
# -*- coding: utf-8 -*-

import os
import datetime
import calendar
import argparse

from PIL import Image, ImageDraw

from inky import InkyPHAT

print("""Inky pHAT: Calendar
Draws a calendar for the current month to your Inky pHAT.
This example uses a sprite sheet of numbers and month names which are
composited over the background in a couple of different ways.
""")

# Get the current path

PATH = os.path.dirname(__file__)

# Command line arguments to set display type and colour

parser = argparse.ArgumentParser()
parser.add_argument('--colour', '-c', type=str, required=True, choices=["red", "black", "yellow"], help="ePaper display colour")
args = parser.parse_args()

colour = args.colour

# Set up the display

inky_display = InkyPHAT(colour)
inky_display.set_border(inky_display.BLACK)

# Uncomment the following if you want to rotate the display 180 degrees
# inky_display.h_flip = True
# inky_display.v_flip = True

def create_mask(source, mask=(inky_display.WHITE, inky_display.BLACK, inky_display.RED)):
    """Create a transparency mask.
    Takes a paletized source image and converts it into a mask
    permitting all the colours supported by Inky pHAT (0, 1, 2)
    or an optional list of allowed colours.
    :param mask: Optional list of Inky pHAT colours to allow.
    """
    mask_image = Image.new("1", source.size)
    w, h = source.size
    for x in range(w):
        for y in range(h):
            p = source.getpixel((x, y))
            if p in mask:
                mask_image.putpixel((x, y), 255)

    return mask_image

def print_digit(position, digit, colour):
    """Print a single digit using the sprite sheet.
    Each number is grabbed from the masked sprite sheet,
    and then used as a mask to paste the desired colour
    onto Inky pHATs image buffer.
    """
    o_x, o_y = position

    num_margin = 2
    num_width = 6
    num_height = 7

[... 104 lines cut ...]
    # And each day in the week
    for col, day in enumerate(week):
        x = (col_w + 1) * col
        x += cal_x + 1

        # Draw in the day name.
        # If it's the current day, invert the calendar background and text
        if (day.day, day.month) == (now.day, now.month):
            draw.rectangle((x, y, x + col_w - 1, y + col_h - 1), fill=inky_display.WHITE)
            print_number((x + 3, y + 3), day.day, inky_display.BLACK)

        # If it's any other day, paint in as white if it's in the current month
        # and red if it's in the previous or next month
        else:
            print_number((x + 3, y + 3), day.day, inky_display.WHITE if day.month == now.month else inky_display.RED)

# Display the completed calendar on Inky pHAT
inky_display.show()

but the output then is: 
Traceback (most recent call last):
  File "cleancal.py", line 114, in <module>
    draw.rectangle((cal_x, cal_y, cal_x + cal_w - 1, cal_y + cal_h - 1), fill=inky_display.BLACK, outline=inky_display.WHITE)
NameError: name 'draw' is not defined
 why is this happening? I mean when the backdrop.png is used there is no error but if i remove the few lines where the backdrop.png is getting loaded it somehow doesn't work out and now i am confused

Gadgetoid, 2020-08-08:
You'll need to put that code paste in between two sets of ``` tags so I can make head or tail of it. Like this:

    ```
    Your code here
    ```

That should prompt GitHub to format it as code rather than markdown.

Some after-image is expected on Inky and can vary depending on ambient temperature and what you're displaying. It ultimately uses the very messy practise of moving ink up and down in a fluid suspension so it's decidedly and unpredictably analog.

kompas1, 2020-08-12:
@Gadgetoid I figured it out I just painted the png full black :sweat_smile: It worked out fine for me. But yes I will from now on change the layout from the code when i post it. Thanks for the tip

kompas1, 2020-08-12:
Oh and I am not sure if i can still put that in but I tried to 'crontab' it for a specific time but if I use cron for executing the command i only get the following error in the log: 

Traceback (most recent call last):
  File "/home/pi/Pimoroni/inky/examples/phat/calendar-phat.py", line 83, in <module>
    text = Image.open("resources/calendar.png")
  File "/usr/lib/python2.7/dist-packages/PIL/Image.py", line 2634, in open
    fp = builtins.open(filename, "rb")
IOError: [Errno 2] No such file or directory: 'resources/calendar.png'

even though it works fine when i enter the directory and run it manually, but not as a cron job

Gadgetoid, 2020-08-13:
For crontab you *must* make sure you `cd` into the directory with `calendar-phat.py` before executing the code.

I usually do this by creating a `start.sh` and adding *that* to crontab instead.

The contents of `start.sh` would look something like:

```
#!/bin/bash
cd /home/pi/Pimoroni/inky/examples/phat/
python calendar-phat.py
```

Now the Python script will run in the proper directory and be able to find its resources.