pimoroni/inky #57: Just doesn't work

cscashby, 2020-08-11:
Two (yellow) Inky Phats, multiple Pi variants (Pi Zero W 1.1, Pi 3B+, Pi 4B)

Main thread here:
https://forums.pimoroni.com/t/inkyphat-not-working/14667/14

Mounting pics here: https://photos.app.goo.gl/RagyP2RHYzHoe6Mw6

If someone can give me some debug help - something like i2cdetect but for SPI, or a way of directly interrogating the board would be handy so I can see what might be wrong here.

Possibly buster specific challenge, not tried with older OS versions.

cscashby, 2020-08-12:
More info on this. I've done a clean stretch install on a Pi Zero W 1.1 (hoping it was a buster specific issue, but no joy).

`apt update`
`apt full-upgrade`

`raspi-config` - enable SPI

Then install:

Complete output attached:
```
pi@raspberrypi:~ $ curl https://get.pimoroni.com/inky | bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 38853  100 38853    0     0   102k      0 --:--:-- --:--:-- --:--:--  103k

This script will install everything needed to use
Inky pHAT & wHAT

Always be careful when running scripts and commands copied
from the internet. Ensure they are from a trusted source.

If you want to see what this script does before running it,
you should run: 'curl https://get.pimoroni.com/inky'

Note: Inky pHAT & wHAT requires SPI communication

Do you wish to continue? [y/N] y

Checking environment...
Updating apt indexes...
...............
...
Checking hardware requirements...

Checking for packages required for GPIO control...
...raspi-gpio is already installed
.....................RPi.GPIO installed and up-to-date

SPI already enabled

Checking packages required by SPI interface...
spidev installed and up-to-date

Inky pHAT & wHAT comes with examples and documentation that you may wish to install.
Performing a full install will ensure those resources are installed,
along with all required dependencies. It may however take a while!

Do you wish to perform a full install? [y/N] y

Checking for dependencies...
python-numpy is already installed
python3-numpy is already installed
python-pil is already installed
python3-pil is already installed
python-lxml is required
Installing python-lxml...
python3-lxml is required
Installing python3-lxml...
python-smbus is already installed
python3-smbus is already installed

Installing Inky pHAT & wHAT library for Python 2...

Collecting inky
  Downloading https://files.pythonhosted.org/packages/88/b9/562fdebbbc0cb6f4781687dd9be31161efb7048b2127c5ec22d847747362/inky-0.0.5-py2-none-any.whl
Collecting smbus2 (from inky)
  Downloading https://www.piwheels.org/simple/smbus2/smbus2-0.3.0-py2.py3-none-any.whl
Requirement already satisfied: numpy in /usr/lib/python2.7/dist-packages (from inky)
Requirement already satisfied: RPi.GPIO in /usr/lib/python2.7/dist-packages (from inky)
[... 271 lines cut ...]
  File "/usr/local/lib/python2.7/dist-packages/smbus2/smbus2.py", line 308, in open
    self.fd = os.open(filepath, os.O_RDWR)
OSError: [Errno 2] No such file or directory: '/dev/i2c-1'
```
now Reboot
```
Linux raspberrypi 4.19.66+ #1253 Thu Aug 15 11:37:30 BST 2019 armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Aug 12 11:54:02 2020 from 192.168.5.103
pi@raspberrypi:~ $ cd Pimoroni/inky/examples/
pi@raspberrypi:~/Pimoroni/inky/examples $ ./name-badge.py --colour yellow --type phat --name "test"
Inky pHAT/wHAT: Hello... my name is:

Use Inky pHAT/wHAT as a personalised name badge!

pi@raspberrypi:~/Pimoroni/inky/examples $ 
```

And display remains stubbornly unchanged.

Gadgetoid, 2020-08-12:
This is *very* likely because you had a new SSD1608-based Inky display and the changes to support these were neither merged nor released- meaning you were running drivers with no idea how to run your shiny new display.

I've rectified this so you should be able to update with:

```
sudo pip install inky --upgrade
sudo pip3 install inky --upgrade
```

Hopefully this will get you up and running. I'm really sorry for all the hassle this caused!

Note: The newer displays have a higher resolution screen and I still haven't ported all the examples over to them.

cscashby, 2020-08-12:
Great stuff - thanks! Am updating now, though the numpy dependency takes a world of time to install on the Pi Zero W - watch this space ;)

Gadgetoid, 2020-08-12:
You might be able to use `sudo apt install python-numpy python3-numpy` to avoid that *long* install. Though I thought numpy had a piwheels package to fix that!

(Ah it's not available for python 2.x - https://www.piwheels.org/project/numpy/)

Gadgetoid, 2020-08-12:
In fact you should use:

```
sudo pip install inky[rpi-gpio-output] --upgrade
sudo pip3 install inky[rpi-gpio-output] --upgrade
```

Because Inky has a non RPi simulator mode now that doesn't require SPI, I2C or GPIO.

I'd better fix that in the one-line installer... somehow!

cscashby, 2020-08-12:
Finally upgraded! --type phat gives the exactly same result; no change on the inky.

I tried --type what (don't think that's right for this) and it complained about resolution
'ValueError: Supplied width/height do not match Inky: 212x104'

So it seems no cigar :(

Is there anything I can do to poke at the electronics / see if the right signals are being presented, etc?

Gadgetoid, 2020-08-12:
You can grab the examples I've recently updated and run them without `--type` since it should auto-detect.

Just:

```
python logo.py
```

Might be that you have old examples!

Gadgetoid, 2020-08-12:
"phat" is the old 212x104 pixel Inky pHAT display, whereas I strongly suspect yours is the new 250x122 pixel Inky pHAT that's named "phatssd1608" in these examples- but just letting auto-detect do its thing is easier than dealing with this byzantine maze of different versions, sizes, colours, etc.

cscashby, 2020-08-12:
Well that was obvious. Sorry I didn't even think. Great stuff. Happy pi geek here!

[image]

closed 2020-08-12 by cscashby

Gadgetoid, 2020-08-12:
HOOOOOOOORAAAAAAAAAAAAAAAAAY!

Hopefully we'll get some time to whip up some new examples that make use of that ever-so-slightly-higher-resolution display. Oh for a @sandyjmacdonald 😆

ljufa, 2020-08-12:
worked for me as well. just make sure to use `-t "auto"` while running examples.

glassworksSean, 2020-08-16:
Hello,
I was having this same issue and my case fits all the symptoms. Assuming that I have one of these newer panels, what would be the basic process from a fresh raspberry pi install? I have tried the steps that are scattered throughout this thread, but I believe I am in a bad state on my pi and the examples don't run.

cscashby, 2020-08-16:
@Gadgetoid 's instructions above worked for me. Note you need to force raspi GPIO mode:
```
sudo pip install inky[rpi-gpio-output] --upgrade
sudo pip3 install inky[rpi-gpio-output] --upgrade
```

Gadgetoid, 2020-08-17:
I'm having second thoughts about `rpi-gpio-output` requirements being a soft dependency- but being able to simulate Inky on the desktop is kinda cool.

@glassworksSean - the one-line-installer *should* work from scratch: `curl -sSL https://get.pimoroni.com/inky | bash`

The examples and library must be in sync for everything to work properly, our one-line-installers usually handle that.

sakumoil, 2021-06-14:
I am having similar kind of problems also. Tried running the `name-badge.py` example:
```python
~/Pimoroni/inky/examples $ python name-badge.py -n"test"
Inky pHAT/wHAT: Hello... my name is:

Use Inky pHAT/wHAT as a personalised name badge!

Detected Black pHAT (SSD1608)
```
but the screen does not update. 
`inky` is at version `1.2.0`. I have enabled SPI over at `raspi-config`.

Gadgetoid, 2021-06-14:
@sakumoil anything else running on your Pi that could be tying up the SPI pins? Since it's write-only it doesn't give any useful error messages when writing just isn't happening at all.

sakumoil, 2021-06-14:
Nothing running on my Pi, it is also a fresh install I am using.

sakumoil, 2021-06-15:
It seems like that my errors were caused by the fact that the inky phat was not correctly seated. I really needed to jam the device into the pins to ensure proper seating.

Gadgetoid, 2021-06-15:
> It seems like that my errors were caused by the fact that the inky phat was not correctly seated. I really needed to jam the device into the pins to ensure proper seating.

This was going to be my next question! It's surprising how much force it needs, and easy to forget once you've been jamming boards on for years.

awdebruijn, 2024-02-24:
Hello,

I'm having the same issues as above. I have the same setup: Pi Zero WH, with a yellow InkyPhat. Installed without any issues, display is properly seated on the pins.
Strange thing is, the display is being auto detected, I can print the colour and resolution. But when I try to show anything on the display, nothing happens... I'm a bit reluctant to disconnect and reseat the ribbon cable because I fear to damage it.

Any ideas what I do wrong or how I can solve this?

awdebruijn, 2024-02-25:
Hi, thanks for the swift response :-)
When I run:
```
from inky.auto import auto
display = auto(verbose = True)
```

The output is:
`Detected Yellow pHAT (SSD1608)`

Is this what you mean?

awdebruijn, 2024-02-26:
@Gadgetoid: I used the Inky on a raspberry with poorly soldered connections before. Right now I have a new pi with pre soldered pins. Is it possible to damage it by doing so?

Gadgetoid, 2024-02-27:
> Is it possible to damage it by doing so?

It's certainly possible, but not very likely. If 5V gets bridged to anything important you'd probably know about it, and Inky's pins are well clear of the 5V supply anyway.

Which examples are you running to try and get the display working?

What version of Raspberry Pi OS?  (`lsb_release -a` and `uname -a` on command-line)

It might also be worth trying the work-in-progress updated code, but only as a last ditch effort since there's probably a simpler answer - https://github.com/pimoroni/inky/pull/182

awdebruijn, 2024-02-28:
I cloned the Inky repo and tried all the examples, no errors, but nothing happens on the screen. 
I'm running Raspberry Pi OS: Raspbian GNU/Linux 11 (bullseye).
The identify.py script outputs a 212x104 yellow, PCB variant 1.2 display variant 12.

awdebruijn, 2024-03-02:
@Gadgetoid: You're probably super busy, but if you don't mind, I want to get to the bottom of this. I bought everything brand new and I can't return it anymore. I would just like to enjoy building cool stuff for this display.
If you could please give me some guidance or another place where I should as for help, I would highly appreciate.

gingertastrophe, 2024-05-22:
I'm having the same issue too with my SSD1608 Red pHAT. It was working fine and the screen refreshed without issues on my Raspberry Pi Zero 2 W, but I had to swap that one out with a new Zero 1 W. I resoldered the pins and now my inky won't update on either of the devices, despite it being detected without issues.