pimoroni/inky #58: Inky pHat red, 1 electrode, EEPROM. Which version is it?

sheepNo, 2020-08-17:
I bough an inky phat, the product page showed a screen with two electrodes. After reading [this blog post](https://blog.pimoroni.com/the-murky-depths-of-inky-phat/) I knew it was the right one to purchase.

Though, when it arrived the screen only had only one electrode. Disappointed, I desperately searched for an hypothetical v3 I hadn't heard about in order to avoid returning my inky.
Eventually I found what I was looking for: "We now populate an EEPROM on each Inky display in order to store information about the type and resolution of each display, but our current approach to reading and acting upon that EEPROM is suboptimal, too." ([issue#53](https://github.com/pimoroni/inky/issues/53))

Also, it works with the new library (this one) and refreshes pretty quickly around 10s instead of 15s as advertised (not complaining here).

Here are pics to show you what I'm talking about:
https://imgur.com/a/jsb53Yo

It's going to go through some intensive use so I'd like to avoid the v1. I'm a bit confused, could you confirm why one electrode and an EEPROM? v1 screen but v3 PCB?

Thanks in advance.

Gadgetoid, 2020-08-17:
That blog post is now some 2-3 years old and so are the displays it deals with- the original "one electrode" Inky pHAT with issues hasn't been in circulation for a long, long time.

You'll most likely have the very latest Inky display which *also* has one electrode... that's... actually quite confusing given the very first display looks similar. If you look closely, however, you will see that the display part is different. The new display also has a different driver integrated into it.

You can double check your board details by running:

```python
from inky.eeprom import read_eeprom
read_eeprom()
```

In a Python repl.

sheepNo, 2020-08-17:
Here is the output:
```python3
Display: 212x104
Color: red
PCB Variant: 1.2
Display Variant: 1
Time: 2019-10-23 09:29:51.7 #Is that manufacturing time ?
```

Yep, if the versions were documented it would avoid some misunderstanding. Why not print it at the back of the PCB?
I've contacted support@pimoroni.com and the reply was along the line of "looks like a v1 to me".

Gadgetoid, 2020-08-17:
Time would be tested time (close enough to manufacturing time), so presumably this is a new-ish Inky but not the new-new Inky.

Keeping track of these display parts has officially broken my brain because I totally missed them switching from 2 whatsits back to 1.

You have what's known internally as a "high temp, Red pHAT" because - and I'm stretching back here trying to remember - suddenly availability of the "low temp" displays vanished and we only had high temp ones. It's... like a new display every year to keep us on our toes!

I will inform support to keep an eye on the EEPROM, and tell them this little trick for identifying boards. Thank you!

sheepNo, 2020-08-17:
I have one last question now, what do you mean by "high temp"? Should I keep an eye on the temperature, anything I should keep in mind while using this screen?

Anyway you've addressed my concerns. I don't think I need to return this one anymore.
Thank you for your quick and detailed answers.

Gadgetoid, 2020-08-17:
It's a reference to the environment the display is designed to function most effectively in. I suspect it means the viscosity of the fluid that the ink particles are suspended in is slightly different. It doesn't make any meaningful difference for you.

Glad you're happy!

closed 2023-01-20 by Gadgetoid