pimoroni/inky #60: Confused about which model I recieved

sjsteer, 2020-09-07:
Hello, I recently purchased the red Inky pHat red (similar to use issue #58) however upon downloading the packages I am faced with a similar issue. There is one electrode on the front of module

[image]

I've attempted to print read_eeprom() but it returns None. 

I honestly have no idea which display has been sent to me and I've contacted the supplier to try and shed some light on the issue.

Gadgetoid, 2020-09-07:
Try running `sudo i2cdetect -y 1` to see if you can see an EEPROM show up. You should see something like this:

```
pi@raspberrypi:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: 50 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

sjsteer, 2020-09-08:
Thank you @Gadgetoid, after checking the connections and reinstalling Rasbpian lite it works.

[image]
[image]

Thank you!

closed 2020-09-08 by sjsteer