pimoroni/inky #61: Can't get it to work on WaveShare 2.13inch e-Paper HAT (B)

hakarune, 2020-09-16:
I'm attempting to use Inky on my new WaveShare 2.13inch e-Paper HAT (B) but nothing shows up, just blank still.

I've installed everything and tried every example, but nadda. Is there a way for this to work? I want to leverage Inky for a pi-hole stats screen and from what I've read, this is very polished system and easy to use...

Using auto mode I get:
```
Traceback (most recent call last):
  File "logo.py", line 44, in <module>
    inky_display = auto()
  File "/home/pi/.local/lib/python3.7/site-packages/inky/auto.py", line 10, in auto
    raise RuntimeError("No EEPROM detected! You must manually initialise your Inky board.")
RuntimeError: No EEPROM detected! You must manually initialise your Inky board.

```

when I run it with the args is says it runs, but no display...
```
pi@raspberrypi:~/Pimoroni/inky/examples$ python3 logo.py --type "phat" --colour "red"
Inky pHAT/wHAT: Logo

Displays the Inky pHAT/wHAT logo.

pi@raspberrypi:~/Pimoroni/inky/examples$
```

Gadgetoid, 2020-09-16:
I'm afraid this library is for Pimoroni Inky products, and we don't make, sell or support WaveShare boards so I'm totally clueless as to what might be going wrong in your case.

As far as I understand it *should* work, but a quick glance over Waveshare's documentation doesn't give me any clues as to which of our drivers are likely to be compatible and doesn't seem to specify which pins their board might use.

I think this is a different display driver part, but here's an example of a WaveShare board pinout: https://pinout.xyz/pinout/213_inch_e_paper_phat

Versus ours: https://pinout.xyz/pinout/inky_phat

hakarune, 2020-09-18:
> I'm afraid this library is for Pimoroni Inky products, and we don't make, sell or support WaveShare boards so I'm totally clueless as to what might be going wrong in your case.
>
> As far as I [...]

Oh, that is unfortunate, I was able to convert the nametag test to work with the waveshare board, but I'm still unable to draw images and after seeing this I was excited, but that is a shame. Back to the drawing board for me. thank you for the prompt reply.

lvan188, 2020-10-25:
Is there a way to solve it?

Gadgetoid, 2020-10-25:
Probably, if you've got a WaveShare board and the time to figure it out!

lvan188, 2020-10-25:
> 也许，如果您有WaveShare板，并且有时间解决这个问题！

Can you solve it?

Gadgetoid, 2020-10-26:
WaveShare provide complete instructions for using their boards on their Wiki - https://www.waveshare.com/wiki/2.13inch_e-Paper_HAT

closed 2020-10-26 by Gadgetoid