pimoroni/inky #63: Fast update for newewr

CattoSalad, 2020-10-08:


CattoSalad, 2020-10-08:
Hello, I purchased an inky phat display (yellow) a few days ago, it seems to be a newer version, I can't seem to figure out a way to have it update faster by only using monochrome.

I was able to identify that my display is an **InkyPHAT_SSD1608**.
I get the following when I read the eeprom:
`Display: 212x104
Color: yellow
PCB Variant: 1.2
Display Variant: 10
Time: b'2020-09-01 10:18:53.8'`

When I use `display = auto()` it detects the resolution as `250x122`, which is what lead me to believe it's an **InkyPHAT_SSD1608**. That and the fact that using `display = InkyPHAT('black'/'yellow')` doesn't seem to work at all.

I have tried `display = InkyPHAT_SSD1608('black')` and it seems to take the same time that it does when I provide it with yellow or use `display = auto()`. Is there a way for me to update the screen faster? Thanks

Gadgetoid, 2020-10-17:
As near as I can tell:

1. The SPI speed can probably be increased for `ssd1608` which at least gets the data transfer faster
2. The `ssd1608` seems to be using internal LUTs so I'll need to poke that until we can:
3. Tweak the pattern of LUTs to effect a faster black/white update- need to understand the format, datasheet here: https://www.displayfuture.com/Display/datasheet/controller/SSD1608.pdf

The TLDR is no, not at the moment. The SPI speed doesn't make a huge difference, since the delay appears to be the actual screen update. I'll make a note to look into this at some point, though, since it looks like- from a glance at other displays running this controller out there- it might be possible to speed up.

AleRoss97, 2020-11-06:
Hello, i'm now intrested in this and i can spend a bit of time doing some modifications in the source for it.
@Gadgetoid did you manage to understand the format of the table at page 22 of the datasheet?

AleRoss97, 2020-11-06:
There seems to be two versions of the inkyphat, i'm using SSD1608 too.

There is another library here for (i think) fast update on the other controller which i'm using to get a grasp on the commands:
https://forums.pimoroni.com/t/my-inky-phat-clock-refresh-speed-question/6955/5

AleRoss97, 2020-11-09:
Re implemented everything (GPIO,SPI,EPAPER lib) in C++ on this repository https://github.com/AleRoss97/epaper .
Changing the LUTS does not seems to have any effect on the update rate as for now.

closed 2023-01-20 by Gadgetoid