pimoroni/inky #78: Mock version of inky doesn't work on windows anymore

Max1412, 2020-11-12:
On a windows desktop, it was possible to use install inky and use inky.mock to simulate an inky display.
Since spidev is a requirement for installing in more recent versions, and spidev is not available on windows, this is not possible anymore.
Is there an option to keep spidev optional or separate the mocking functionality/library from the actual implementation which needs spidev?

Gadgetoid, 2020-11-16:
The big issue here is that dropping the necessary requirements for Raspberry Pi made setting up this library for 99.9% of users a complete and total pain- we ended up with a bunch of confused customers who had non-working installs because the dependencies weren't handled.

Making this even more of a pain- there doesn't seem to be an easy way to include dependencies (for Pi) and then optionally exclude them (for Windows).

I think the simple option is to just use pips `--no-dependencies` switch, ie: 

```
python3 -m pip install inky --no-dependencies
```

This gets Inky installed on Windows for me, and the simulator just works.

Max1412, 2020-11-17:
Using `--no-dependencies` solves this for me as described (and I can totally understand that providing a smooth install experience on the pi matters most). Thank you!