pimoroni/inky #80: Library seemingly no longer works with a stock Raspbian install

Maffsie, 2020-11-16:
I'm trying to get started with my Inky Impression display, and have flashed the Raspbian Minimal latest image using the Raspberry Pi flasher, then ran `curl https://get.pimoroni.com/inky | bash` per the instructions on the store page, choosing to install the examples and documentation.
Upon doing so, rebooting, and manually enabling I2C as the script didn't actually enable it somehow, rebooting again and trying to run any of the example scripts other than the buttons and identify scripts fails.

When using Python2 and running `from inky.auto import auto; disp=auto()` I do correctly get an inky_uc8159 object, but when I then call `disp.setup()` (or indeed any other method that writes to the SPI bus) I get a `TypeError: Non-Int/Long value in arguments` exception.
When using Python3 and attempting to run `from inky.auto import auto` I get a `UnicodeDecodeError`.

This happens reproducibly with the latest Raspbian Minimal and the setup script, using both the Impression and my existing Inky wHAT (for which this library worked the last time I used it around a year ago), on multiple Pi 3 boards.

Gadgetoid, 2020-11-16:
This is odd- any idea where those exceptions are occuring?

Additionally what version of Python 3? - `python3 --version`

Maffsie, 2020-11-16:
Hi, I managed to sort it out by only using Python 3 (it's 3.8), in addition to uninstalling the bumpy packages provided by Raspbian - it appears that both the python2 and python3 packages are broken, as installing numpy from pip worked (after installing a dependency that I forget the name of at present).
It's working now, but it's definitely not clear whether I should've been using py2 or py3, and it may be worthwhile updating the install script to bypass attempting to install from the Apt repos (at least for numpy), and adding shebangs to the example scripts to indicate they're for python3 specifically.

closed 2022-01-13 by helgibbons