pimoroni/inky #87: pip3 installs old version

bablokb, 2020-12-21:
pip3 seems to install an old version of the Inky-Impression code (set_border still raises an exception).

ahnlak, 2021-01-29:
This has been fixed in https://github.com/pimoroni/inky/pull/82 but that hasn't yet made it into the 'official' release, so `pip3` won't pick it up by default.

However, you can work around it by cloning this repo and installing it from there:

`pip3 install /home/pi/inky/library/`

(assuming you've cloned the repo into `/home/pi`, obviously adapt the path as required!)

Gadgetoid, 2021-01-29:
Just released v1.2.0 which should hopefully sort this!

bablokb, 2021-02-05:
Thanks, pip now supplies the updated version. We can close this issue.

closed 2021-02-05 by bablokb