pimoroni/inky #92: inky.set_border() from examples/7color/cycle.py is not implemented for  inky_uc8159

byako, 2021-01-17:
Here is how it fails if you try to run "cycle" example for 7color (Inky Impression):
```
~/git/inky/examples/7color $ python3 cycle.py
Color: Black
Traceback (most recent call last):
  File "cycle.py", line 16, in <module>
    inky.set_border(color)
  File "/home/tango/.local/lib/python3.7/site-packages/inky/inky_uc8159.py", line 369, in set_border
    raise NotImplementedError
NotImplementedError
```

ahnlak, 2021-01-29:
This has been fixed in https://github.com/pimoroni/inky/pull/82 but that hasn't yet made it into the 'official' release, so `pip3` won't pick it up by default.

However, you can work around it by cloning this repo and installing it from there:

`pip3 install /home/pi/inky/library/`

(assuming you've cloned the repo into `/home/pi`, obviously adapt the path as required!)

byako, 2021-01-29:
Great, thanks @ahnlak for notifying, and to @joe-warren for the fix :+1:

closed 2021-01-29 by byako