pimoroni/inky #96: Commandline image conversion

peteruithoven, 2021-01-31:
I'd love a quicker way to convert images to the supported palette. Ideally also in a way that eliminates as many possible mistakes as possible. 

I've used inky-palette.gpl but didn't get it to work. I had to edit and export the existing backdrop.png. I also want to convert a lot of weather icon images. 

Is there any way to execute this conversion through gimp through the command line? (I'm on Linux so any bash, zsh compatible command would work)
If this is possible it becomes very easy to convert multiple files.

peteruithoven, 2021-01-31:
Interesting: https://www.gimp.org/tutorials/Basic_Batch/

The command `gimp-context-set-palette` seems what we want? 

```
(define (inky filename)
   (let* ((image (car (gimp-file-load RUN-NONINTERACTIVE filename filename)))
          (drawable (car (gimp-image-get-active-layer image))))
     (gimp-context-set-palette "inky")
     (gimp-file-save RUN-NONINTERACTIVE image drawable filename filename)
     (gimp-image-delete image)))
```

In my case I've got Gimp installed as flatpak so I had to save this to:
`~/.var/app/org.gimp.GIMP/config/GIMP/2.10/scripts`

And normally you'd run it like:
```
gimp -i -b '(inky "icon.png")' -b '(gimp-quit 0)'
```

Because if flatpak I had to run it like: 
```
/usr/bin/flatpak run --branch=stable --arch=x86_64 --command=gimp-2.10 --file-forwarding org.gimp.GIMP -i -b '(inky "icon.png")' -b '(gimp-quit 0)'
```

Does not yet work though...

peteruithoven, 2021-01-31:
`gimp-image-convert-indexed` seems what I needed. 

```
(define (inky filename)
   (let* ((image (car (gimp-file-load RUN-NONINTERACTIVE filename filename)))
          (drawable (car (gimp-image-get-active-layer image))))
     (gimp-image-convert-indexed
        image
        CONVERT-DITHER-NONE
        CONVERT-PALETTE-CUSTOM
        3
        FALSE
        TRUE
        "inky")
     (gimp-file-save RUN-NONINTERACTIVE image drawable filename filename)
     (gimp-image-delete image)))
```
Save this script as `inky.scm` in the GIMP scripts folder. 
Add the palette to gimp with the name "inky". 

Execute by running:
```
gimp -i -b '(inky "file.png")' -b '(gimp-quit 0)'
```

This script clearly simplifies the colors into the 3 colors.
But white becomes black, the palette seems off...

bablokb, 2021-02-05:
Why don't you just copy (and modify) the python-code from the Inky-Impression class to a small python-script?

closed 2023-01-20 by Gadgetoid