pimoroni/inky #97: Auto works, but manual doesn't

peteruithoven, 2021-02-01:
I can't get the my Yellow pHAT (SSD1608) to render in manual mode. 
Simple test: 

``` python3
from PIL import Image, ImageFont, ImageDraw
from font_hanken_grotesk import HankenGroteskBold
from inky.auto import auto
from inky import InkyPHAT

try:
    inky_display = auto(ask_user=True, verbose=True)
    # inky_display = InkyPHAT('black') # Nothing happens
    # inky_display = InkyPHAT('yellow') # Nothing happens
except TypeError:
    raise TypeError("You need to update the Inky library to >= v1.1.0")

hanken_bold_font = ImageFont.truetype(HankenGroteskBold, 40)

img = Image.new("P", inky_display.resolution)
draw = ImageDraw.Draw(img)
draw.text((0, 0), "Test", inky_display.BLACK, font=hanken_bold_font)

inky_display.set_image(img)
inky_display.show()
```

I was hoping to test the theory that manually initializing my yellow pHAT as black would increase the refresh rate.

peteruithoven, 2021-02-08:
I had a look at [auto's source](https://github.com/pimoroni/inky/blob/master/library/inky/auto.py) and the verbose output auto gave me:
```
Detected Yellow pHAT (SSD1608)
```
This invormed me that I needed to use `InkyPHAT_SSD1608` instead of `InkyPHAT`. Which worked :) 

Full example working for my Inky
``` python3
from PIL import Image, ImageFont, ImageDraw
from font_hanken_grotesk import HankenGroteskBold
from inky.auto import auto
from inky import InkyPHAT

try:
    inky_display = InkyPHAT_SSD1608('black')
except TypeError:
    raise TypeError("You need to update the Inky library to >= v1.1.0")

hanken_bold_font = ImageFont.truetype(HankenGroteskBold, 40)

img = Image.new("P", inky_display.resolution)
draw = ImageDraw.Draw(img)
draw.text((0, 0), "Test", inky_display.BLACK, font=hanken_bold_font)

inky_display.set_image(img)
inky_display.show()
```

closed 2021-02-08 by peteruithoven

Gadgetoid, 2021-02-08:
Glad you've got this up and running- sorry I didn't jump in! Been busy filling endless cracks with decorator's caulk and breathing sawdust.

peteruithoven, 2021-02-08:
Ha no problem, good luck with the home improvements Gadgetoid