pimoroni/inky #100: Cannot set border with newer inkyphat (SSD1608)

keillor, 2021-02-16:
When I run the command set to control the border on one of the new inkyphats, nothing happens. I believe this is due to the resolution size being slightly different (250x122 instead of 212x104).

jerbzz, 2021-02-25:
I also always get a white border on my yellow Inky pHat, no matter what I set, such as: `inky_display.set_border(inky_display.YELLOW)`

`Found: Yellow pHAT (SSD1608)
Display: 212x104
Color: yellow
PCB Variant: 1.2
Display Variant: 12
Time: b'2020-10-08 17:17:35.9'
`

jerbzz, 2021-02-26:
@Gadgetoid any thoughts on next steps I could take for debugging? I'm not a Python or hardware expert by any means but certainly happy to get my hands dirty.

jerbzz, 2021-03-01:
OK - so in `inky_ssd1608.py` the border wasn't getting set at all. 

I've now added this (with some modifications) from `inky.py`...

```        if self.border_colour == self.BLACK:
            print('Setting black border...')
            self._send_command(ssd1608.WRITE_BORDER, 0b00000000)  # GS Transition Define A + VSS + LUT0
        elif self.border_colour == self.RED and self.colour == 'red':
            print('Setting red border...')
            self._send_command(ssd1608.WRITE_BORDER, 0b01110011)  # Fix Level Define A + VSH2 + LUT3
        elif self.border_colour == self.YELLOW and self.colour == 'yellow':
            print('Setting yellow border...')
            self._send_command(ssd1608.WRITE_BORDER, 0b00110011)  # GS Transition Define A + VSH2 + LUT3
        elif self.border_colour == self.WHITE:
            print('Setting white border...')
            self._send_command(ssd1608.WRITE_BORDER, 0b00110001)  # GS Transition Define A + VSH2 + LUT1
```
and I can now make it black or white on my yellow pHAT... trying to set yellow makes a slightly different colour but not proper yellow... hang in there. Anyone want to send me a red SSD1608 pHAT to try? :P

jerbzz, 2021-03-30:
I'm planning to submit a PR for this but I am really struggling to get my hands on a red SSD1608 Inky pHAT! The Pi Hut and Amazon both sent me old style ones, as did Pimoroni after assuring me they only had new style ones in stock!

Gadgetoid, 2021-03-30:
> I'm planning to submit a PR for this but I am really struggling to get my hands on a red SSD1608 Inky pHAT! The Pi Hut and Amazon both sent me old style ones, as did Pimoroni after assuring me they only had new style ones in stock!

Sorry for the lack of response here, I've been WFH and having to focus down on some select projects to avoid being overwhelmed.

I'll check up regarding an SSD1608-based red Inky pHAT. I'm not 100% sure if they exist and could have sworn we only had yellow. Not disambiguating between old/new products with their own SKUs was a mistake.

I *really* appreciate the effort you're putting in here- I hope it hasn't left you too out of pocket!? :grimacing:

jerbzz, 2021-03-30:
Nah it's fine, I was just hoping for a matching pair of red and yellow! I've returned the non-SSD1608 ones from Amazon/Pi Hut, but still waiting to hear back from Pimoroni Support about why I got a low res when they said it would be a high res! 

I know everyone is busy :)

If the red SSD1608 doesn't exist then I will have to just suck up the fact that my displays won't match - scaling factors argh!!

chorltonwheelies, 2021-03-30:
Not sure if this helps (I will check using the Inky eeprom function), but
as the owner of a black and also a red Inky PHAT, I can report that the red
unit is definitely a lower resolution compared to the former: I have had to
adjust all draw positions initially set for the black variety (bought in
mid 2020), to make the same python script display correctly on the red
(bought January 2021).

jerbzz, 2021-03-30:
I heard from Hel at Pimoroni support this afternoon:

"We've solved the mystery :)

Turns out we only started manufacturing red Inky pHATs with SSD1608 screens on 19 March, which is why you would have received one of the old ones. We've had the screens since the end of last year so I think everyone assumed they would have filtered into the system by now!"

One of them is on the way to me - hot off the production line yesterday!

jerbzz, 2021-04-08:
Fixed with https://github.com/pimoroni/inky/pull/108

Gadgetoid, 2021-07-26:
Should be released as of - https://pypi.org/project/inky/

Thanks all!

closed 2021-07-26 by Gadgetoid