pimoroni/inky #110: Pwnagotchi help ??? not displaying on phat v2 model # FPC-7528

lildaze, 2021-04-08:
can y'all check it out and see if anything wrong thanks

main.name = "pwnagotmon"
main.lang = "en"
main.whitelist = [
  "Pretty Fly for"
]

main.plugins.grid.enabled = true
main.plugins.grid.report = true
main.plugins.grid.exclude = [
  "Pretty Fly for"
]

ui.display.enabled = true
ui.display.type = "inky"
ui.display.color = "black"

lildaze, 2021-04-08:
I can ssh in to it I did the systemctl status pwnagotchi then I get this error

pwnagotchi.service - pwnagotchi Deep Reinforcement Learning instrumenting bett
   Loaded: loaded (/etc/systemd/system/pwnagotchi.service; enabled; vendor prese
   Active: activating (auto-restart) (Result: exit-code) since Wed 2019-07-10 02
     Docs: https://pwnagotchi.ai
  Process: 341 ExecStart=/usr/bin/pwnagotchi-launcher (code=exited, status=1/FAI
 Main PID: 341 (code=exited, status=1/FAILURE)

stets, 2021-04-09:
This is probably better suited in the [pwnagotchi repo](https://github.com/evilsocket/pwnagotchi)

Did you get the color parameter right?

it looks like the systemd service is restarting repeatedly. Can you pull logs or try to start the pwnagotchi service from the command line and see what happens? Get the start command from `/etc/systemd/system/pwnagotchi.service`

lildaze, 2021-04-09:
ok I just posted it on the Pwnagotchi repo take a look have the logs on there too ran the service then I get this , color parameter ? I think so my inkys a black n white model 

==== AUTHENTICATION FAILED ===
Failed to start pwnagotchi.service: Access denied

Gadgetoid, 2021-04-09:
If you have the Inky library installed you can dump the ID eeprom with: `python3 -m inky.eeprom` which might help.

The display variant is the best way to tell boards apart:

1. Red pHAT
2. Yellow wHAT
3. Black wHAT
4. Black pHAT
5. Yellow pHAT
6. Red wHAT
7. Red wHAT (High Temp)
8. Red wHAT
9. N/A
10. Black pHAT (SSD1608 version)
11. Red pHAT (SSD1608 version)
12. Yellow pHAT (SSD1608 version)
13. N/A
14. 7-Colour

Note: most of our pHAT range has now been replaced by the SSD1608 variants - 10, 11 and 12 - which still erroneously report as 212x104 but are actually 250x122 "pixels". They use an SSD1608 driver which I raised an issue about here - https://github.com/evilsocket/pwnagotchi/issues/941 - but I'm unsure if support has been integrated beyond the posted zip file.

lildaze, 2021-04-09:
hey thanks man I did the dump and got exactly what you said , yeah I seen the issue on evilsocket/pwnagotchi/issues/941 
now I don't really know what to do , any recommendation on installing the drivers ?

pi@pwnagotchi:~ $ python3 -m inky.eeprom
/usr/lib/python3.7/runpy.py:125: RuntimeWarning: 'inky.eeprom' found in sys.modules after import of package 'inky', but prior to execution of 'inky.eeprom'; this may result in unpredictable behaviour
  warn(RuntimeWarning(msg))
Display: 212x104
Color: black
PCB Variant: 1.2
Display Variant: 10
Time: b'2020-10-08 16:32:22.8'

Gadgetoid, 2021-04-09:
I suggested a patch here that might work, but I've got no time/space to set up the whole pwngotchi stack and test it yet - https://github.com/evilsocket/pwnagotchi/issues/980

A little creative modification to your pwngotchi install should be able to apply that patch if you're willing to be a test subject?

Gadgetoid, 2021-04-09:
I've pushed up to a fork, which you can grab in zip form from here: https://github.com/Gadgetoid/pwnagotchi/archive/refs/heads/patch-inky-ssd1608.zip and substitute in the pwnagotchi step-by-step install instructions or- perhaps?- install over your existing installation.

You need to configure `ui.display.color` to "auto" ... I have zero experience with pwnagotchi though so I'm shooting in the dark a little here.

lildaze, 2021-04-09:
oh ok thank you for your time ill see if I can figure this out I'm new to this kind of stuff n still learning , ok ill check it out thanks again gadetoid .

Gadgetoid, 2021-04-09:
No worries! I should have been more proactive about getting a fix worked out for this- but... *gestures vaguely at current situation*

Made the first steps to fix this in pwnagotchi itself, but it's very untested :grimacing:  - https://github.com/evilsocket/pwnagotchi/pull/985

ahaltric, 2021-04-23:
check my post on https://github.com/evilsocket/pwnagotchi/issues/979

closed 2023-01-20 by Gadgetoid