pimoroni/inky #111: Ellipse not drawing solid outline

PeaceDealer, 2021-04-21:
I'm using the inky wHAT, and I'm trying to draw an analog clock face, however, the outline of the PIL Ellipse tool doesn't seem to be solid. Got lots of white dots sprinkled around the outline.

This is the code which I'm using
```
inky_display = InkyWHAT('red')
inky_display.h_flip = True
inky_display.v_flip = True

colourBackground = inky_display.WHITE
colourForeground = inky_display.BLACK
colourSpecial = inky_display.RED

inky_display.set_border(colourForeground)

img = Image.new("P", (inky_display.WIDTH, inky_display.HEIGHT), colourBackground)

draw = ImageDraw.Draw(img)
draw.ellipse((20, 20, 280, 280), fill = None, outline = colourForeground, width=5)

inky_display.set_image(img)
inky_display.show()
```

jerbzz, 2021-04-21:
Can you show us a picture of what you get?

PeaceDealer, 2021-04-21:
Like this

[image]

jerbzz, 2021-04-21:
How interesting. It looks like it's maybe drawing 5 concentric ellipses and then aliasing is happening on the parts which aren't straight?

Do you get a solid line if you specify width = 1?

PeaceDealer, 2021-04-21:
Its a continues line, however not very circular. There are places where it seems a little jittery, which could seem to line up with the white spots at width 5.

I guess this may be an issue with the PIL library then.

I was just trying to generate an RGB image, where its not an issue, but I cant get the display to process that directly.

Gadgetoid, 2021-04-21:
Testing this in pure PIL on my desktop yields a solid - if someone ropey - line as you might expect:

[image]

Code:

```python
from PIL import Image, ImageDraw, __version__

print(__version__)

image = Image.new("P", (300, 300), 0)

draw = ImageDraw.Draw(image)

draw.ellipse((20, 20, 280, 280), fill=None, outline=1, width=5)

image.putpalette([255, 255, 255, 0, 0, 0])

image = image.resize((600, 600), Image.NEAREST)

image.show()
```

Assuming there wasn't some weird version of PIL where circles weren't rendered solid, I'm guessing `set_image` is failing to detect the image as "P" (palette) type and trying to quantize it.

But in the interest of science, what does the above code do for you?

PeaceDealer, 2021-04-21:
This is the exact code i ran on the raspberry pi:
```
from PIL import Image, ImageDraw, __version__
print(__version__)
image = Image.new("P", (300, 300), 0)
draw = ImageDraw.Draw(image)
draw.ellipse((20, 20, 280, 280), fill=None, outline=1, width=5)
image.putpalette([255, 255, 255, 0, 0, 0])
image = image.resize((600, 600), Image.NEAREST)
image.save("test.png")
```

and the image that it generates looks like this:
[image]

So it seems the PIL libary for some reason is to blame. It returns as version `5.4.1` on the pi.

jerbzz, 2021-04-21:
[image]

Same on my Inky pHAT with PIL 5.4.1.

Gadgetoid, 2021-04-21:
Well since PIL is intent on being.... weird... (even my circles leave much to be desired) ... perhaps this will work better:

```python3
from PIL import Image, ImageDraw, __version__

print(__version__)

image = Image.new("P", (300, 300), 0)

draw = ImageDraw.Draw(image)

width = 5

draw.ellipse((20, 20, 280, 280), fill=1, outline=None)
draw.ellipse((20 + width, 20 + width, 280 - width, 280 - width), fill=0, outline=None)

image.putpalette([255, 255, 255, 0, 0, 0])

image = image.resize((600, 600), Image.NEAREST)

image.show()
```

jerbzz, 2021-04-21:
I was on the stock inky library install using the bash script from the product page. 

I just:

sudo apt remove python3-pil

sudo pip3 install -Iv pillow==6.2.2
(latest version that supports Python 2.7)

sudo apt install libopenjp2-7

and now:

[image]

PeaceDealer, 2021-04-22:
Can confirm, after changing to the pillow Libary everything works fine.