pimoroni/inky #112: Supplied width/height do not match Inky - 7-Colour (UC8159)

henneaux, 2021-05-02:
Hi, 
The inky library throws a ValueError as soon I'm trying to initiate the inky_display. Some examples are working just fine, but most examples return "ValueError: Supplied width/height do not match Inky: 600x448". 

As soon "inky_display = InkyWHAT("red")" is defined, the error is raised:
```
pi@raspberrypi-inky:~/Pimoroni/inky/examples $ python3
Python 3.7.3 (default, Jan 22 2021, 20:04:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from inky import InkyWHAT
>>> inky_display = InkyWHAT("red")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/inky/what.py", line 44, in __init__
    v_flip=False)
  File "/usr/local/lib/python3.7/dist-packages/inky/inky.py", line 92, in __init__
    raise ValueError('Supplied width/height do not match Inky: {}x{}'.format(self.eeprom.width, self.eeprom.height))
ValueError: Supplied width/height do not match Inky: 600x448

```

example logo.py (works - logo is drawn on the Inky):
```
pi@raspberrypi-inky:~/Pimoroni/inky/examples $ python3 logo.py 
Inky pHAT/wHAT: Logo

Displays the Inky pHAT/wHAT logo.

Detected 7-Colour (UC8159)
```

Identify.py:
```
pi@raspberrypi-inky:~/Pimoroni/inky/examples $ sudo python3 identify.py
Found: 7-Colour (UC8159)
Display: 600x448
Color: None
PCB Variant: 1.2
Display Variant: 14
Time: b'2021-02-08 11:17:38.8'
```

Am I missing something obvious? Thank you.

jerbzz, 2021-05-02:
According to identify.py, you have an Inky Impression, not an Inky wHAT... can you show a picture of the display you have?

henneaux, 2021-05-02:
Hi jerbzz, Thanks. 
Correct it says 'pimoroni.com/impression' on the back (link to product: https://shop.pimoroni.com/products/inky-impression).

Does this mean the Inky impression is not compatible/supported with the inky-library (= pHAT/wHAT only) ?

jerbzz, 2021-05-02:
Just use 

    from inky.auto import auto
    board = auto()

You are telling the library you have an Inky wHAT by stating

henneaux, 2021-05-02:
Yes, thanks for your help. I understand now!

closed 2021-05-02 by henneaux