pimoroni/inky #113: Inky Impression 7 colour - No EEPROM detected

Krandelbord, 2021-05-11:
I have Inky Impression (7 colour ePaper/eInk/EPD)
(https://thepihut.com/products/inky-impression-7-colour-epaper-eink-epd)

When I do according to instructions:
`
from inky.auto import auto
board = auto()
`
I get this tracback. 

> Traceback (most recent call last):
>   File "<pyshell>", line 1, in <module>
>   File "/home/pi/.local/lib/python3.7/site-packages/inky/auto.py", line 44, in auto
>     raise RuntimeError("No EEPROM detected! You must manually initialise your Inky board.")
> RuntimeError: No EEPROM detected! You must manually initialise your Inky board.

jerbzz, 2021-05-11:
> File "", line 1, in

It looks like the inky is not being detected. How have you got it hooked up?

Krandelbord, 2021-05-12:
It is connect by GPIO

Daveyr, 2021-08-08:
How did you fix this? I have a yellow phat and a black phat and both are throwing the same errors as yours. They are directly mounted to a Pi4 8Gb.

Krandelbord, 2021-08-09:
instead of auto I used "from inky import Inky7Colour as Inky"
So to render some image it looks like this:
```
from inky import Inky7Colour as Inky
from PIL import Image, ImageDraw
inkyphat = Inky()
inkyphat.set_image(rendered_image)
inkyphat.show()

```

georgeHall, 2022-11-06:
HI @Krandelbord, Not sure if you have got anywhere with this. I had a similar issue and found the issue to be the I2C channels being used. 

I have a Pi4 and was using the 64 bit RPI os, the I2C bus names are different which means the line https://github.com/pimoroni/inky/blob/master/library/inky/eeprom.py#L137 which is supposed to use the bus `/dev/i2c-1`. On the 64 bit os the names are `/dev/i2c-20` and `/dev/i2c-21`, this means it cannot find the eeprom as its not on the i2c bus.

I have made a manual fix in my code to change the line `i2c_bus = SMBus(1)` to `i2c_bus = SMBus(20)`

darkmavis1980, 2022-11-14:
> HI @Krandelbord, Not sure if you have got anywhere with this. I had a similar issue and found the issue to be the I2C channels being used.
>
> I have a Pi4 and was using the 64 bit RPI os, the I2C bus [...]

Hi @georgeHall, I'm finding myself in the same issue, I applied that change, and while I don't get an error anymore, the little script, which is literally the one to print your name [found on pimoroni](https://learn.pimoroni.com/article/getting-started-with-inky-phat), does nothing 🤔

georgeHall, 2022-11-14:
Hi @darkmavis1980, we would need to do some debugging to get some more information. 

It might be easiest if we chat in the Discord server: https://discord.gg/xZP76WYy

closed 2023-01-20 by Gadgetoid