pimoroni/inky #115: How to show an image on Inky Impression?

callumlocke, 2021-07-01:
The example image is a `.jfif` file. I want to render a regular format like a PNG to the Inky Imression. I have tried to piece it together from examples and snippets found online but I can't get it working. Are there any special rules for exporting or generating 7-colour images for this? I can't find any documentation about this.

callumlocke, 2021-07-01:
Trying to figure things out from the source code, came across something strange - in the [code here](https://github.com/pimoroni/inky/blob/fa59811bb5a9aad9eac82da3e58b6dc2ead75b8c/library/inky/inky_uc8159.py#L26-L46) in `inky_uc8159.py`, the names `SATURATED_PALETTE` and `DESATURATED_PALETTE` seem to be backwards.

For example the 'red' in the `SATURATED_PALETTE` is `156, 72, 75` and in `DESATURATED_PALETTE` it's `255, 0, 0`.

Edited to show swatches in IDE:

[image]

callumlocke, 2021-07-09:
I managed to render a PNG (exported from Photoshop as a 24-bit RGB PNG, with the image only containing pixels in the exact 7 colours that the Inky Impression supports), but the colours are all randomly mixed up on the display. I also tried exporting as an 8-bit indexed PNG from Photoshop, but couldn't get this working at all.

Just to clarify what I'm asking, here's a PNG I made in Photoshop:

[image]

It is the right size (600x448), and it contains only 7 colours, the exact ones the Inky Impression supports.

How can I display an image like this on the Inky Impression, with the colours in the right order?

jerbzz, 2021-07-09:
The order of channels is critical. I used The GIMP to get it to work
properly. It has a pretty decent custom palette feature.

Gadgetoid, 2021-07-27:
> I don't really understand why there's 'saturated' and 'desaturated' sets, with 14 colours between them. Can the display actually show 14 colours? I would really appreciate some input from Pimoroni to help me understand and get off the ground.

These colours represent the values Inky will quantize toward when converting and dithering a full colour image for display on the 7-colour board.

The `SATURATED` palette includes de-saturated colours which - when used as the target for quantization - produce a saturated output image on the display.

The `DESATURATED` palette is the inverse of this. By pretending the display's colours are super saturated the resulting image usually appears less saturated.

What *actually* happens internally is that we take these two palettes and mix them together to provide a target with a customizable amount of saturation.

Our final palette - the mix of saturated/desaturated colours - is the target toward which all pixels in the image you're displaying will be clamped. When you use a desaturated target (that's much less vibrant than the actual display colours) then desaturated colours in the source image are more likely to be snapped toward a more vibrant display colour (or dithered mix of colours, as it were) than they otherwise would. And the reverse is also true.

So TLDR- these variables are named for the result they produce, and not for the colours they contain. The colours were obtained by photographing the display - for the saturated palette - and using the full-saturation colours - for the desaturated palette.

Inky will have a good go at displaying basically any image you throw at it, using these colours as a target for quantization/dithering.

For example here's a particularly challenging source image:

[image]

Fully saturated:

[image]

Fully desaturated:

[image]

While the desaturated image is quantized toward how the colours actually *look* on Inky Impression the result is pretty underwhelming because the bright colours don't match up well with what's actually possible on the display.

The desaturated image actually has truer-to-the-original brightness and dynamic range, so there are very clearly some tradeoffs.

Here's the same image with a saturation of 0.7 which is bringing back some of the dynamic range at the expense of losing the popping colours/vibrancy:

[image]

This is all only applicable to automatically dithering photographs for display on Inky- if you're creating your own art targeting the colours it supports and outputting a palette-based image then - as @jerbzz suggests - it's only the *colour order* that matters and not the actual colours since the raw image data is just copied verbatim into Inky's display buffer.

KodeMunkie, 2022-05-04:
I've written an improved general use algorithm (non PIL based) for dithering, based on testing with various dithers and colour distance algorithms based on my existing work in other projects.
The results seem to be better for the poor sky-blue output on these devices, and also avoids over saturation of yellow found in the advanced dithers included in the project.
Unfortunately there doesn't seem to be a method to bypass the "blend" saturation function and dump the image directly onto the display.

Is there a way to do this cleanly?

I attach a screenshot of the above image run through my algorithm choices as a preview. Nb. this does not use the exact sat or desat palettes given in the source but an adapted palette, taken from a real sample done by another user of the generic waveshare 7 color variant, that produces a more pleasing result actually on the device (e.g. blues are not greens,  off white contains white+slight blue tints, not white+yellow etc.).
[image]
.

Gadgetoid, 2022-06-06:
If you supply a "P" type PIL image - which should be the output from your dither operation - then no conversion will take place the data is copied verbatim to the display.

Conversion only happens if the image is not already a palette type image:

https://github.com/pimoroni/inky/blob/45024c0b5cd2475ef7c2c96d9f22af2890c1fed5/library/inky/inky_uc8159.py#L385-L395

KodeMunkie, 2022-06-06:
Thanks for pointing out the above.

Currently I provide a 24 bit image that has been quantized already to 7 "idealistic" colours. The colours need a simple 1:1 mapping to the official physical hardware palette for the device. I am uncertain as to which constants, sat or desat I need to use (I think it's the sat, but would like confirmation).

This operation should be as simple as a simple 1:1 palette mapping by index, but try as I might I can't find the convert function at line 395 documented or precisely what it does.
I am also unable to use the PIL documentation image.convert() (which has different args) as it _requires_ an optional matrix, which given lack of adequate documentation I can only assume is a dither matrix. This confuses me as the only dither option I would use is "None" so I don't trust it to not try and do some sort of redundant processing or further dither.

I realise this isn't the PIL project but was wondering if you had some pointers as to how best to map the 7 colour RGB palette to a 7 colour P image without the convert function, or at least point me to the line 395 convert function documentation/source you used?

Should you be interested the python source for the diffuse image conversion is a relatively small 100 line file here: https://github.com/KodeMunkie/inky-impression-slideshow/blob/main/image_processor.py

closed 2023-01-20 by Gadgetoid

turian, 2024-01-30:
I would be interested to learn more about this too, @KodeMunkie

KodeMunkie, 2024-02-01:
@kyleratti @turian Probably not what you're looking for but the algorithm I wrote is in a (greatly modified by me) fork of another project: https://github.com/KodeMunkie/inky-impression-slideshow