pimoroni/inky #117: Inky Yellow pHAT (SSD1608) Screen Won't Update

aelamrn1, 2021-07-10:
Hi. I recently bought a yellow Inky pHAT (SSD1608) and I'm using a Raspberry Pi Zero Wireless (RPZ). After installing the latest Raspberry Pi OS, I connected to the RPZ via SSH and ran the following one-liner to install the latest inky pHAT software.
`curl https://get.pimoroni.com/inky | bash`

I then made sure SPI was enabled with `raspi-config`.

I also ran the following:
`sudo pip install inky[rpi-gpio-output] --upgrade`
`sudo pip3 install inky[rpi-gpio-output] --upgrade`

Currently the Inky pHAT is being detected, since running the name-badge.py example now gives
`Inky pHAT/wHAT: Hello... my name is:`
`Use Inky pHAT/wHAT as a personalised name badge!`
`Detected Yellow pHAT (SSD1608)`

However, the board _refuses_ to update, and instead stubbornly remains on the factory default screen. I can't think of anything else to do, and unfortunately, I'm pretty sure I've read every similar issue raised, but with no luck.

Any ideas?

Note: I rebooted several times at all stages to make sure any changes properly propagate.

Gadgetoid, 2021-07-27:
The SSD1608 can be identified - I think, anyway - by the return of *two* electrode connections on the right-hand side.

The resolution difference you're seeing is normal- the wrong values were flashed to EEPROM due to the 250x122 resolution actually coming as something of a surprise. The class accounts for this.

If you haven't already contacted support, it might be worth doing since this could be a hardware issue.

lesamouraipourpre, 2021-08-16:
I'm getting the same issue with a Yellow pHat (received Sat 14th August)
```
./identify.py 
Found: Yellow pHAT (SSD1608)
Display: 212x104
Color: yellow
PCB Variant: 1.2
Display Variant: 12
Time: b'2021-06-03 11:24:40.4'
```
It correctly reads the EEPROM over I2C but appears to get no response over SPI - the screen is still on the factory shipped screen.

aelamrn1, 2021-08-16:
> I'm getting the same issue with a Yellow pHat (received Sat 14th August)
>
> ```
> ./identify.py
> Found: Yellow pHAT (SSD1608)
> Display: 212x104
> Color: yellow
> PCB Variant: 1.2
> Display Variant: 12
> Time: [...]

aelamrn1, 2021-08-16:
> I'm getting the same issue with a Yellow pHat (received Sat 14th August)
>
> ```
> ./identify.py
> Found: Yellow pHAT (SSD1608)
> Display: 212x104
> Color: yellow
> PCB Variant: 1.2
> Display Variant: 12
> Time: [...]

Yeah it's really frustrating. I'm going to contact Pimoroni directly because I've made no progress figuring it out.

Gadgetoid, 2021-08-17:
Asking someone to pull one from stock and investigate this, since I seem to have a shortage of Inky pHATs here.

lesamouraipourpre, 2021-08-18:
It looks like my issue was a local hardware issue. Because of the case on my Raspberry Pi I have to use a GPIO extender. I tried swapping it for another extender that I have and it seems to be working fine. I can't find a continuity fault in the extender but I'm blaming it anyway.

aelamrn1, 2021-08-18:
After contacting support, Pimoroni offered to exchange my Inky with a new one. Works fine now! must have been a hardware issue. Anyway, great support software support and customer service.

closed 2022-01-13 by helgibbons