pimoroni/inky #118: `clean.py` doesn't work on Inky pHAT no matter what you do

raspberrycoulis, 2021-07-19:
I believe I have an early model Inky pHAT (Yellow), but trying to run the `clean.py` example to clear my screen does not work, no matter what I try.

Running `./clean.py -t phat -c yellow` results in:

```bash
Inky pHAT: Clean

Displays solid blocks of red, black, and white to clean the Inky pHAT
display of any ghosting.

Failed to detect an Inky board, you must specify the type and colour manually.

usage: clean.py [-h] [--number NUMBER]
clean.py: error: unrecognized arguments: -t phat -c yellow
```

Based on the above, it looks like the argument `-n x` is required, so tried again with `./clean.py -t phat -c yellow -n 1` but get the exact same error message.

However, running other examples - such as `name-badge.py` with the same `-t phat -c yellow` works just fine, but the script still complains:

```bash
Inky pHAT/wHAT: Hello... my name is:

Use Inky pHAT/wHAT as a personalised name badge!

Failed to detect an Inky board, you must specify the type and colour manually.
```

raspberrycoulis, 2021-07-19:
There are also many installation issues when running `curl https://get.pimoroni.com/inky | bash`, such as:

```bash
This script will install everything needed to use
Inky pHAT & wHAT

Always be careful when running scripts and commands copied
from the internet. Ensure they are from a trusted source.

If you want to see what this script does before running it,
you should run: 'curl https://get.pimoroni.com/inky'

Note: Inky pHAT & wHAT requires SPI communication
Note: Inky pHAT & wHAT requires I2C communication

Do you wish to continue? [y/N] y

Checking environment...
Updating apt indexes...
...............
Installing python-pip...
Extracting templates from packages: 100%
.....................Installing python3-pip...
........................
Checking hardware requirements...

Checking for packages required for GPIO control...
raspi-gpio is already installed
.....................RPi.GPIO installed and up-to-date

SPI must be enabled for Inky pHAT & wHAT to work
SPI is now enabled

Checking packages required by SPI interface...
spidev installed and up-to-date

I2C must be enabled for Inky pHAT & wHAT to work
I2C is now enabled

Checking packages required by I2C interface...
Python 2 smbus installed and up-to-date
Python 3 smbus installed and up-to-date

Inky pHAT & wHAT comes with examples and documentation that you may wish to install.
Performing a full install will ensure those resources are installed,
along with all required dependencies. It may however take a while!

Do you wish to perform a full install? [y/N] y

Checking for dependencies...
python-numpy is required
Installing python-numpy...
python3-numpy is required
Installing python3-numpy...
python-pil is required
Installing python-pil...
python3-pil is required
Installing python3-pil...
python-lxml is required
Installing python-lxml...
python3-lxml is required
Installing python3-lxml...
python-smbus is already installed
python3-smbus is already installed

Installing Inky pHAT & wHAT library for Python 2...

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting inky
  Downloading https://files.pythonhosted.org/packages/86/da/824b373c86698cc4211183d57a3723aaf7636a56e905b2bfe1c9ea72b69a/inky-1.2.0.tar.gz
[... 275 lines cut ...]
Apt failed to install python3-wikiquotes!
Falling back on pypi...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting wikiquotes
  Downloading https://www.piwheels.org/simple/wikiquotes/wikiquotes-1.4-py3-none-any.whl
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from wikiquotes) (2.21.0)
Collecting bs4 (from wikiquotes)
  Downloading https://www.piwheels.org/simple/bs4/bs4-0.0.1-py3-none-any.whl
Collecting unidecode (from wikiquotes)
  Using cached https://files.pythonhosted.org/packages/9e/25/723487ca2a52ebcee88a34d7d1f5a4b80b793f179ee0f62d5371938dfa01/Unidecode-1.2.0-py2.py3-none-any.whl
Requirement already satisfied: lxml in /usr/lib/python3/dist-packages (from wikiquotes) (4.3.2)
Collecting slacker (from wikiquotes)
  Downloading https://www.piwheels.org/simple/slacker/slacker-0.14.0-py3-none-any.whl
Requirement already satisfied: beautifulsoup4 in /usr/lib/python3/dist-packages (from bs4->wikiquotes) (4.7.1)
Installing collected packages: bs4, unidecode, slacker, wikiquotes
Successfully installed bs4-0.0.1 slacker-0.14.0 unidecode-1.2.0 wikiquotes-1.4

Downloading examples and documentation...
Resources for your Inky pHAT & wHAT were copied to
/home/pi/Pimoroni/inky

Finalising Install...

Some packages could not be installed, review the output for details!
```

Gadgetoid, 2021-07-27:
The install issues are normal, we prefer apt packages where available and fall back to pypi- this ends up being a bit noisy, eg:

```
Installing python3-wikiquotes...
E: Unable to locate package python3-wikiquotes
Apt failed to install python3-wikiquotes!
Falling back on pypi...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting wikiquotes
  Downloading https://www.piwheels.org/simple/wikiquotes/wikiquotes-1.4-py3-none-any.whl
  ```

As for `clean.py` it looks like `parse_args` needs to change to `parse_known_args` on this line: https://github.com/pimoroni/inky/blob/e073c8e5d1bffa27f370c0c6ae1ecccabb62a366/examples/clean.py#L22

That's a bug!

raspberrycoulis, 2021-08-02:
Just adding - I have changed the line mentioned above and this now works as expected.

PR #121 (2021-08-02): Add mocking support to auto()
Add the ability to mock WHAT, PHAT, PHATSSD1608 and Impression by supplying the `--simulate` command-line argument.

Fix a bug with clean.py where it tried to parse the auto() args and bailed. Fixes #118
closed 2021-08-02 by Gadgetoid