pimoroni/inky #119: trying to use Pimoroni 7 color 5.4in with RPi0+ and having issues

sniffski, 2021-07-20:
---
:~/inky/examples/7color# ./clear.py 
Traceback (most recent call last):
  File "./clear.py", line 7, in <module>
    inky = Inky()
  File "/usr/local/lib/python3.7/dist-packages/inky-1.2.0-py3.7.egg/inky/inky_uc8159.py", line 137, in __init__
    self.eeprom = eeprom.read_eeprom(i2c_bus=i2c_bus)
TypeError: read_eeprom() got an unexpected keyword argument 'i2c_bus'
---
:~/inky/examples/7color# ./colour-palette.py 
Traceback (most recent call last):
  File "./colour-palette.py", line 17, in <module>
    inky = Inky()
  File "/usr/local/lib/python3.7/dist-packages/inky-1.2.0-py3.7.egg/inky/inky_uc8159.py", line 137, in __init__
    self.eeprom = eeprom.read_eeprom(i2c_bus=i2c_bus)
TypeError: read_eeprom() got an unexpected keyword argument 'i2c_bus'

None of the examples work for me... :( any ideas?

Gadgetoid, 2021-07-27:
Something has gone drastically wrong with your library install, since `read_eeprom` *does* accept an `i2c_bus` keyword argument- https://github.com/pimoroni/inky/blob/c5743f9fcfe9c2d6863ceb770895f2cbaa6fc1cd/library/inky/eeprom.py#L121-L133

I've just released a couple of new versions, so maybe a `python3 -m pip install inky --upgrade` will fix this.

sniffski, 2021-07-29:
> Something has gone drastically wrong with your library install, since `read_eeprom` *does* accept an `i2c_bus` keyword argument- [...]

Looks like you are right! Thanks allot!