pimoroni/inky #122: Is there a "good" way to only update changed pixels?

cyrlop, 2021-08-17:
Is there a way to speed-up the refresh process by only changing specific pixels instead of going through the whole blinking/flashing cycle?

I use my Inky phat (black) to display a big number at the center, updating every 6 to 20 seconds, it works well so far.

There a couple of solutions around but none that I can figure out how to translate to my inky phat black:
- [This guy on the Pimoroni forum did explained it well](https://forums.pimoroni.com/t/my-inky-phat-clock-refresh-speed-question/6955/5)
- [This one changed timing for pwnagotchi project](https://github.com/evilsocket/pwnagotchi/blob/e9899dda94b7015af0204552e6a5d862a4bbf651/pwnagotchi/ui/inkyphat/inkyfast.py)

**In summary: is the only way to optimize refresh to mess with timings risking damaging the screen or is there a cleaner way if only few pixels change?**

Gadgetoid, 2021-08-30:
I’ve done a lot of messing with the timings and haven’t damaged a screen yet.

Some displays do support updating a region of the display, but I’m not sure any of ours do.

Most of the blinking/flashing is an attempt to erase the previous image from the screen and avoid image retention- you can drastically shorten this period and then run a periodic full update to clean up the screen. If you’ve ever used an e-reader you might notice them doing this.

cyrlop, 2021-08-30:
Thanks for the clarification!
The issue is I have no idea how to do it. I have the "ssd1608" screen (black).
I know I'm supposed to [change some timings at this line](https://github.com/pimoroni/inky/blob/master/library/inky/inky_ssd1608.py#L106) but have no idea about the values to remove the black and white flashing.

I plan to display the new image as fast as possible and run a full cycle (or clean.py) from times to times.

Thanks again for the help!

tilusnet, 2022-05-12:
is there any similar hack at hand for the UC8159 chip?

Gadgetoid, 2022-05-12:
> is there any similar hack at hand for the UC8159 chip?

Nope. There is no partial update support, nor even support for custom LUTs.

closed 2023-01-20 by Gadgetoid