pimoroni/inky #136: Inky Impression - EEPROM Error

phatmanace, 2021-12-23:
I wondered if anyone might have some ideas on this problem?

I have the inky impression 5.7" display and I can't get it to work. I have it hooked up to a pi zero w, ran the installer script from the pimoroni website, and I get this below. EEPROM error sounds like it could be a faulty unit to me, but I don't have a good way of proving otherwise. 

```
pi@rp-mini-phat:~/Pimoroni/inky/examples $ sudo python3 clean.py 
Inky pHAT: Clean

Displays solid blocks of red, black, and white to clean the Inky pHAT
display of any ghosting.

Detected 7-Colour (UC8159)
Traceback (most recent call last):
  File "/home/pi/Pimoroni/inky/examples/clean.py", line 17, in <module>
    inky_display = auto(ask_user=True, verbose=True)
  File "/usr/local/lib/python3.9/dist-packages/inky/auto.py", line 24, in auto
    return InkyUC8159()
  File "/usr/local/lib/python3.9/dist-packages/inky/inky_uc8159.py", line 144, in __init__
    eeprom_resolution = _RESOLUTION.keys[self.eeprom.display_variant - 14]
TypeError: 'builtin_function_or_method' object is not subscriptable
pi@rp-mini-phat:~/Pimoroni/inky/examples $ 

```

... and the system Diag

```

pi@rp-mini-phat:~/Pimoroni/inky/examples $ uname -a
Linux rp-mini-phat 5.10.63+ #1459 Wed Oct 6 16:40:27 BST 2021 armv6l GNU/Linux
pi@rp-mini-phat:~/Pimoroni/inky/examples $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@rp-mini-phat:~/Pimoroni/inky/examples $ 

```

helgibbons, 2021-12-24:
I'm getting this error with the 5.7 Inky and the latest version of the library too - I think there might be a bug in the new bit of the UC8159 driver that tries to tell the difference between the 4" and 5.7" screen?

As a temporary fix (until @Gadgetoid returns from remote, Inky-less lands), you could try replacing `/usr/local/lib/python3.9/dist-packages/inky/inky_uc8159.py` with the version in the [previous release](https://github.com/pimoroni/inky/releases/tag/v1.2.2)?

phatmanace, 2021-12-24:
Nice - that worked - Thank you for the suggestion; the display now works!

.. now the next problem I'm getting is a timeout waiting for the busy signal to clear..time to go and ferret around in the library!

```
etected 7-Colour (UC8159)
Traceback (most recent call last):
  File "/home/pi/Pimoroni/inky/examples/name-badge.py", line 107, in <module>
    inky_display.show()
  File "/usr/local/lib/python3.9/dist-packages/inky/inky_uc8159.py", line 365, in show
    self._update(buf.astype('uint8').tolist(), busy_wait=busy_wait)
  File "/usr/local/lib/python3.9/dist-packages/inky/inky_uc8159.py", line 329, in _update
    self._busy_wait()
  File "/usr/local/lib/python3.9/dist-packages/inky/inky_uc8159.py", line 309, in _busy_wait
    raise RuntimeError("Timeout waiting for busy signal to clear.")
RuntimeError: Timeout waiting for busy signal to clear.

```

phatmanace, 2021-12-24:
... which I've now sorted out. Just changed the timeout to 60s. Happy Holidays!

> Nice - that worked - Thank you for the suggestion; the display now works!
>
> .. now the next problem I'm getting is a timeout waiting for the busy signal to clear..time to go and ferret around in the [...]

helgibbons, 2021-12-24:
Glad that sorted things temporarily!

Leaving this issue open as we still need to fix it properly (also the lurking `busy_wait` issues)

Gadgetoid, 2021-12-24:
Hopefully this PR should fix the detection issues and prevent any from creeping in - https://github.com/pimoroni/inky/pull/137

I should probably merge and publish since anything will be better than the current embarrassingly broken mess.

pdsavas, 2021-12-27:
I tried replacing the inky_uc8159.py file with the one from version 1.22 (as per the link), but I am still getting the Traceback error. I have a fresh install of Buster, and used the curl command to install the inky software. The version of Python that Buster installed is 3.7, and the original poster noted he was running 3.9. I'm really new to all of this.

What should I do?

Thanks in advance.

phatmanace, 2021-12-27:
> I tried replacing the inky_uc8159.py file with the one from version 1.22 (as per the link), but I am still getting the Traceback error. I have a fresh install of Buster, and used the curl command to [...]

Just to be clear, you executed the steps in the following order?

- Installed the software with the bash/curl command from the Pimoroni site as per instructions. 
- Replace the single file ```/usr/local/lib/python3.9/dist-packages/inky/inky_uc8159.py``` with the one from 1.22?

If you are and it’s still not working, can you post the error message here, and also the version of python and the OS you are running?

pdsavas, 2021-12-27:
Yes. That is what I did. This is the error I get when I run cycle.py in the Thonny Python IDE:

Traceback (most recent call last):
  File "/home/pi/Pimoroni/inky/examples/7color/cycle.py", line 16, in <module>
    inky.show()
  File "/usr/local/lib/python3.7/dist-packages/inky/inky_uc8159.py", line 365, in show
    self.update(buf.astype('uint8').tolist(), busy_wait-busy_wait)
  File "/usr/local/lib/python3.7/dist-packages/inky=inky_uc8159.py", line 226, in setup
    self._busy_wait()
  File "/usr/local/lib/python3.7/dist-packages/inky/inky_uc8159.py", line 309, in _busy_wait
    False RuntimeError("Timeout waiting for busy signal to clear.")
RuntimeError: Timeout waiting for busy signal to clear.

As I mentioned, I am running this on a fresh install of Buster (full with desktop), and Python 3.7.

Thanks in advance.

phatmanace, 2021-12-27:
> Yes. That is what I did. This is the error I get when I run cycle.py in the Thonny Python IDE:
> 
> Traceback (most recent call last): File "/home/pi/Pimoroni/inky/examples/7color/cycle.py", line 16, in inky.show() File "/usr/local/lib/python3.7/dist-packages/inky/inky_uc8159.py", line 365, in show self.update(buf.astype('uint8').tolist(), busy_wait-busy_wait) File "/usr/local/lib/python3.7/dist-packages/inky=inky_uc8159.py", line 226, in setup self._busy_wait() File "/usr/local/lib/python3.7/dist-packages/inky/inky_uc8159.py", line 309, in _busy_wait False RuntimeError("Timeout waiting for busy signal to clear.") RuntimeError: Timeout waiting for busy signal to clear.
> 
> As I mentioned, I am running this on a fresh install of Buster (full with desktop), and Python 3.7.
> 
> Thanks in advance.

I think that's the other problem, the timeout issue. 

If you look at [This file](https://github.com/pimoroni/inky/pull/137/commits/2fd500e3138a33a3b85fd6e86de3959d4aa67521) - you should be able to see where to make the relevant change.

pdsavas, 2021-12-27:
It didn't work. I changed that line to "def _busy-wait(self, timeout=6-,0):" and got this error:

Traceback (most recent call last):
  File "/home/pi/Pimoroni/inky/examples/7color/cycle.py", line 16, in <module>
    inky.show()
  File "/usr/local/lib/python3.7/dist-packages/inky/inky_uc8159.py", line 365, in show
    self._update(buf.astype('uint8').tolist(), busy_wait-busy_wait)
  File "/ust/local/lib/python3.7/dist-packages/inky/inky_uc8159.py", line 226, in setup
    self-_busy_wait()
  File "/ust/local/lib/python3.7/dist-packages/inky/inky_uc8159.py", line 309, in _busy_wait
    False RuntimeError("Timeout waiting for busy signal to clear.")
RuntimeError: Timeout waiting for busy signal to clear.

It looks identical to the one before I made that change. (sigh)

phatmanace, 2021-12-27:
Did you really mean ```6-```` (dash/hyphen)? - think it should be ```60```

pdsavas, 2021-12-27:
I meant "60;" I fat-fingered it. :)

pdsavas, 2021-12-27:
[image]

phatmanace, 2021-12-27:
Sorry then - those were the only two changes I made and I got mine working... Alas you'll need to debug a bit further. 

one other nugget of information was that when I fixed issue number 1 (the other/old version of the librarry) - everything started working, the screen actually did refresh, i just needed to extend the timeout. Is your screen actually doing anything?

pdsavas, 2021-12-27:
No, it's not doing anything. :(

phatmanace, 2021-12-27:
Can you share a bit more about your set up - which exact hardware (inky + also the Raspberry pi you're using) - will see if there is anything obvious I can spot

pdsavas, 2021-12-27:
It's a Raspberry Pi 3B+ with an Inky Impression. I'm using a ribbon cable to connect the Pi to the Inky. What else would you like to know?

jerbzz, 2021-12-27:
Here is a really silly question - but are you certain you have the display
the correct way round on the ribbon cable?

pdsavas, 2021-12-27:
Yes; I double-checked that several times in fact for that very reason.

helgibbons, 2022-01-12:
I'm closing this issue as the original EEPROM error is now resolved in the most recent release. I've tested it with Bullseye and 4" and 5.7" screens and both types are now being detected correctly.

@pdsavas - If you're still having trouble with timeout errors it might be worth trying running the examples direct from the terminal rather than from Thonny to see if that improves things:

```
cd ~/Pimoroni/inky/examples/7color/
python3 stripes.py
```

Might also be worth double checking you've got I2C and SPI enabled in `sudo raspi-config`, and trying without the ribbon cable in case it's making a bad connection with the busy pin?

closed 2022-01-12 by helgibbons

pdsavas, 2022-01-13:
So do I essentially need to do a fresh setup with Bullseye to get this working? I've heard that Bullseye is still buggy.

helgibbons, 2022-01-13:
The Inky library should work OK with Buster as well, but starting from scratch with a fresh Raspberry Pi OS image now that the library is updated is probably a good idea.

pdsavas, 2022-01-15:
You're going to laugh (I hope). I went back and looked to see whether I2C and SPI were enabled on the Pi. I2C was not; SPI was. Once I enabled I2C, everything started working.

Now I just need to find a good slideshow program to use with this thing. :)

Thanks for the help!

helgibbons, 2022-01-17:
Excellent, glad to hear you got it going :)