pimoroni/inky #139: Inky impression: driver install script.

phatmanace, 2022-01-05:
Wondered if any of the maintainers here owned the install script for the inky impression drivers

[Link here](https://shop.pimoroni.com/products/inky-impression-5-7)

Specifically, this script

```curl https://get.pimoroni.com/inky | bash```

I noticed a minor annoyance with it that it wants to use ```apt-get ``` to install all the python libraries that the driver needs, but has to fail back for each library to the ```pip``` method - not a massive deal, but makes the script take ages to run as it has to wait for a timeout for each one. Would also be happy to be told that I’m using the wrong OS on my RPI, although it was a fresh install of Raspbian. 

didn’t know if @Gadgetoid  happened to own this per chance.

Feel free to close if I need to pursue another place for discussing this.

Gadgetoid, 2022-01-05:
These nightmarish scripts are a very lightly maintained legacy holdover that I've been slowly supplanting with an `install.sh` in each individual repo.

If you peer under the hood you'll understand why I'm apprehensive to change them.

phatmanace, 2022-01-05:
No problem, I’ll just stay as I am and put with the other script for now; Thanks a lot for responding though!

Gadgetoid, 2022-02-14:
> Would also be happy to be told that I’m using the wrong OS on my RPI, although it was a fresh install of Raspbian.

This- at least- should now be fixed.

closed 2023-03-03 by ageeandakeyboard