pimoroni/inky #142: wHat: 90 degree rotation

wisemoth, 2022-04-04:
I can set `self.rotation` to be `0` and `180` without issue.

However if I set to `90` or `270` the display is garbled.  It looks like there is library code to support 90 degree rotation, but am I expected to flip `width` and `height` in my own code or? Are there examples?

gleich, 2022-05-28:
I am having the same issue with the phat. @wisemoth did you do anything to fix this?

wisemoth, 2022-07-05:
@gleich I'm afraid not - I ended up drawing the UI in landscape mode, but would still prefer to run in portrait mode if that became possible.

Gadgetoid, 2022-07-05:
Inky just uses a Pillow `Image` to draw the UI, you can draw into a portrait image and rotate that to landscape before sending it to `display.set_image()`

IE:

```python
from PIL import Image

img = Image.new("P", (300, 400))

inky.set_image(img.rotate(90, expand=True))
```