pimoroni/inky #148: 404 for libtiff5 package needed for examples that use PIL

thnewlands, 2022-08-11:
I was getting this missing package error when I tried to run the examples after installing the latest version of Pillow on my Rasberry Pi.
```
  File "logo.py", line 4, in <module>
    from PIL import Image
  File "/home/pi/.local/lib/python3.7/site-packages/PIL/Image.py", line 100, in <module>
    from . import _imaging as core
ImportError: libtiff.so.5: cannot open shared object file: No such file or directory
```

I read elsewhere that running ```sudo apt install libtiff5``` would install the missing package. So I tried to run it and I found this 404 error instead:
```
sudo apt install libtiff5
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  libtiff5
0 upgraded, 1 newly installed, 0 to remove and 116 not upgraded.
Need to get 250 kB of archives.
After this operation, 594 kB of additional disk space will be used.
Err:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf libtiff5 armhf 4.1.0+git191117-2~deb10u2
  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/t/tiff/libtiff5_4.1.0+git191117-2~deb10u2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Does anyone know of a mirror that works for libtiff5? Am I installing Pillow incorrectly? 

It might be useful to include a section that explains how to get the proper components of Pillow necessary to run the library in the readme.

thnewlands, 2022-08-11:
Thank you for following up so quickly! 

```
sudo apt install python3-pil
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  libimagequant0 liblcms2-2 libtiff5 libwebpdemux2 libwebpmux3 python3-olefile
Suggested packages:
  liblcms2-utils python-pil-doc python3-pil-dbg
The following NEW packages will be installed:
  libimagequant0 liblcms2-2 libtiff5 libwebpdemux2 libwebpmux3 python3-olefile python3-pil
0 upgraded, 7 newly installed, 0 to remove and 116 not upgraded.
Need to get 614 kB/974 kB of archives.
After this operation, 2,712 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Err:1 http://raspbian.raspberrypi.org/raspbian buster/main armhf libtiff5 armhf 4.1.0+git191117-2~deb10u2
  404  Not Found [IP: 93.93.128.193 80]
Err:2 http://raspbian.raspberrypi.org/raspbian buster/main armhf python3-pil armhf 5.4.1-2+deb10u2
  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/t/tiff/libtiff5_4.1.0+git191117-2~deb10u2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/p/pillow/python3-pil_5.4.1-2+deb10u2_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

Strangely I'm getting a similar set of missing IPs here. I don't think it's entirely on my end because I've been able to download plenty of other packages and have run ```sudo apt update``` several times

Gadgetoid, 2022-08-17:
I'm not sure what's going on here, but I think your package management is subtly broken.

Those broken URLs are broken for me too, but the package manager shouldn't even be trying to download stuff that doesn't exist.

A reinstall might be necessary since you'll likely continue to have trouble installing other things (this issue has nothing directly to do with our libraries), or perhaps try:

```
sudo apt update
sudo apt upgrade
sudo apt --fix-missing update
```

thnewlands, 2022-09-11:
That fixed my problem!! Thank you.

closed 2022-09-11 by thnewlands