pimoroni/inky #154: weather example stopped working

KayleaS, 2023-01-10:
the weather example for phat stopped working for me as of new years eve 2022

Inky pHAT: Weather

Displays weather information for a given location. The default location is Sheffield-on-Sea.

Detected Red pHAT (SSD1608)
Traceback (most recent call last):
  File "/home/pi/Pimoroni/inky/examples/phat/weather-phat.py", line 118, in <module>
    weather = get_weather(location_string)
  File "/home/pi/Pimoroni/inky/examples/phat/weather-phat.py", line 82, in get_weather
    weather["summary"] = curr[0].img["alt"].split()[0]
IndexError: list index out of range

helgibbons, 2023-01-10:
Looks like Apple discontinued the Dark Sky API (which is how this example gets its weather data) at the end of 2022 :( https://techcrunch.com/2022/09/13/as-apples-weatherkit-launches-dark-sky-for-ios-to-wind-down-operations-by-year-end

There's a list of some alternative weather APIs here, though I think most of them will need you to sign up for an API key to get at their data: https://medium.com/@Ari_n/8-weather-api-alternatives-now-that-darksky-is-shutting-down-42a5ac395f93

We've used Open Meteo for some of our Raspberry Pi Pico examples which doesn't require an API key - might be worth a look! https://open-meteo.com/

helgibbons, 2023-01-17:
Updated example here if you want to give it a try :)

https://github.com/pimoroni/inky/blob/patch/remove-darksky/examples/phat/weather-phat.py

PR #156 (2023-01-20): Use Open Meteo API instead of Dark Sky
Update weather example to get its data from Open Meteo instead of the now defunct Dark Sky. We have wind speed instead of atmospheric pressure now, for tidiness of API request reasons :)

Fixes #154
closed 2023-01-20 by Gadgetoid