pimoroni/inky #155: Feature request: faster monochrome updates on Inky Impression

joel0, 2023-01-14:
Would it be possible to do less comprehensive display refreshing to display black & white while skipping all other 5 colors?

My use case updates the display every 5 minutes and only needs color on rare occasions.  I use a large font for viewing from a distance, so I wouldn't even mind if it leads to some visible artifacts.

I'm not sure whether this is technically possible with the Impression hardware, but seeing how many other e-ink products on the market have options to tradeoff speed for quality, I figured it's worth asking at least.

helgibbons, 2023-01-16:
We've not found a way to do this with the seven colour displays I'm afraid - as I understand it the screen driver chip on these displays mandates a full clear cycle every time so we're not able to modify the refresh behaviour.

closed 2023-01-20 by Gadgetoid

joel0, 2023-01-22:
Thank you for taking the time to answer my question. :)