pimoroni/inky #162: How to map palette to impression 7.3 colours and avoid dithering

ReX1983, 2023-04-01:
Hello, 

I have a Home Assistant dashboard that I want to show on the inky impression 7.3".
I get the snapshot via headless chromium, then convert it to 800x480, but now I am facing a dithering issue and the final result is low quality.

The snapshot has just 9 colours:
- light blue
- red
- violet
- dark yellow
- green
- black
- dark grey
- white
- light grey

I am not after colour accuracy, what I am hoping to achieve is to translate the above colour to the native colours of the impression screen, something like:
- light blue > solid blue 
- red > red
- violet > orange
- dark yellow > yellow
- green > green
- black > black
- dark grey > black
- white > white
- light grey > white

Is there a way to achieve this in either python or command line?

Thanks!

orestesgaolin, 2025-05-13:
Wondering if you managed to solve it?

ReX1983, 2025-05-14:
> Wondering if you managed to solve it?

No and kind of gave up on my project (home automation dashboard), as the screen refresh is too slow and not as readable as I hoped.

orestesgaolin, 2025-05-14:
Yeah, I'm facing similar issue now. The time it takes to refresh is sometimes inconvenient. Even if I make it refresh every 30-60 min it may happen that when I look at the screen it will be mid-refreshing and it will take 20 more seconds for the image to appear.

Enverex, 2025-11-15:
You *should* be able to do something like this with ImageMagick's mapping feature, you provide a swatch of colours and it will convert the image to use only those, you can also disable dithering to get a more direct image.

But for some reason I'm getting a black background when this is sent to the display, despite it being pure white (255,255,255) in the image.