pimoroni/inky #163: Make inky work with NVIDIA Jetson Nano

andreicimpean, 2023-04-15:
Looks like I was able to set everything to work from a software point of view on Jetson Nano but there is a warning:
--> Jetson.GPIO ignores setup()'s pull_up_down parameter
Is this something that is preventing the use of inky or there could be another configuration issue that does this?

Gadgetoid, 2023-06-13:
That sounds suspiciously like Jetson is using libgpio under the hood, with no support for pull up/down. When designing for the Pi we generally go by the assumption that it's possible to set board pull-ups or pull-downs, and thus don't populate resistors to handle this. You could populate an external resistor to handle this. It wouldn't be pretty, but it should work.

Gadgetoid, 2024-10-01:
This may or may not be fixed with the latest release of Inky, which migrates us to libgpiod/gpiod (which now supports BIAS setting, though whether Jetson Nano has a recent enough kernel.. I know not!)

Anyway to be clear this is a "it might actually work now" rather than an "it's officially supported" since we don't have or use Jetson Nano or really any SBCs other than Pi (too many boards, too little time!)

Good luck!

closed 2024-10-01 by Gadgetoid