pimoroni/inky #172: Use on any non-Pi machines?

DanielArnett, 2023-08-01:
Does anyone know if this driver can be used on any non-Pi devices? If I have something similar that has SPI pins (like a beaglebone or jetson) is it able to connect to the inky?

Thank you,

SpencerWF, 2023-11-10:
There should be any reason you can't use it on any device, but you will need to reverse-engineer the communication that's happening in the pi libraries as I don't know if anyone has done this yet.

Gadgetoid, 2023-11-10:
Migrating this library from RPi.GPIO over to gpiod - which is on my TODO list - should very much help with that. The spidev/smbus libraries are pretty standard already, but Pi GPIO has been the exception.

DanielArnett, 2023-11-10:
Thank you both for the feedback. I don't have an immediate need for this (Pi ended up being fine for my project), but it might one day be necessary. If I end up having to do this I'll make sure to submit a PR here.

Gadgetoid, 2024-10-01:
gpiod support is now merged :partying_face: 

I don't actually *have* any alternate boards to test on, but in theory the driver should now work anywhere that has gpiod.

closed 2024-10-01 by Gadgetoid