pimoroni/inky #174: Busy Wait error - Line 323/333

KRuddPM, 2023-09-01:
Had my Inky Impressions working for weeks, and today I've loaded the script up and I'm getting the following error:

```
/usr/local/lib/python3.9/dist-packages/inky/inky_uc8159.py:323: UserWarning: Busy Wait: Held high. Waiting for 0.20s
  warnings.warn("Busy Wait: Held high. Waiting for {:0.2f}s".format(timeout))
/usr/local/lib/python3.9/dist-packages/inky/inky_uc8159.py:333: UserWarning: Busy Wait: Timed out after 0.20s
  warnings.warn("Busy Wait: Timed out after {:0.2f}s".format(time.time() - t_start))
```

I have looked through the Issues here, and tried adjusting the timeout on the inky_uc8159.py script, but I still get the same error.

I have the Inky connected directly to the PIs GPIO pins, with no other IO.

Any help would be appreciated.

SamKingDev, 2023-10-11:
Did you manage to get anywhere with this?

Animeefan, 2024-11-07:
Any progress on this issue or a related issue?

admga, 2025-01-02:
Same issue here, using a Raspberry Pi 4.

Animeefan, 2025-01-02:
So, recently the issue was resolved for me by changing the RPi board. Originally used a Zero 2W but then switched to a Pi4 and later to a different Zero 2W. Worked on both of them. I suppose there was likely a short on my gpio that caused the busy wait error.

kedare, 2025-02-08:
I have the same issue on a Zero 2W too, I don´t think it's a short issue, more likely performance bottleneck?

Animeefan, 2025-02-08:
> I have the same issue on a Zero 2W too, I don´t think it's a short issue, more likely performance bottleneck?

No I’m certain the reason was a short, I tried with a second Zero 2W and it worked. I also found the short on the gpio pins. I still do get the busy wait error but it uploads just fine, so I recommend you just try it with another rpi too.