pimoroni/inky #177: error: externally-managed-environment

ChopsKingsland, 2023-10-17:
I can't seem to install this package.

I've tried both manually doing `pip3 install` and using `install.sh`, and both give the same error...

[image]

I also tried `pip3 install` with `--break-system-packages` and it failed for a different reason...

Gadgetoid, 2023-10-17:
Read the error message 😆

Also, see this absolute wall of text I've been cobbling together about this problem and how we might fix it via our install scripts: https://github.com/pimoroni/boilerplate-python/pull/13

Now, if you'd join me in an anguished scream... 😭

helgibbons, 2024-09-11:
Closing this one as we've now merged the new installer into main - it should now attempt to set up a virtual environment for you, if you don't have one.

There's updated install instructions in the readme: https://github.com/pimoroni/inky/blob/main/README.md

I've also updated our Inky pHAT and wHAT Learn Guides:
https://learn.pimoroni.com/article/getting-started-with-inky-phat
https://learn.pimoroni.com/article/getting-started-with-inky-what

closed 2024-09-11 by helgibbons